parallaxinc/Parallax-IDE #10: BlocklyProp Support

DarkmatterVale, 2015-02-23:
Add support for BlocklyProp. I do not know if this is possible.

PropGit, 2015-03-09:
This is definitely in our plans, but for a later phase.  We have a lot of ground to cover first before we can start absorbing functionality such as this.

DarkmatterVale, 2015-03-09:
Definitely. We don't even have a release for BlocklyProp yet, and that has to be completed first before anything else can be done.

PropGit, 2015-03-26:
Ken responded to my query about supporting BASIC Stamp with Blockly with the following, which I interpret as meaning "No... at least, not right now, but we may readdress it again later."  We do however have Propeller support with Blockly, so we may later choose only to bring in Blocky as a language for Propeller in our ChromeIDE.

From Ken:
For the BASIC Stamp? I don’t think so - the PAB WX makes it a lot more interesting. 

I would’ve never thought about BASIC Stamp support with Blockly, but it’s an interesting idea. I’m sure that in 1998 we would’ve done anything to have such a tool for education.