parallaxinc/Parallax-IDE #9: Blink program for BS2

phated, 2015-02-23:


PropGit, 2015-02-23:
Here's a simple Blink LED program for the BASIC Stamp 2 (BS2).  It also outputs a DEBUG stream.
*[GitHub won't let me attach the file, so I've just entered it here, but you'll see that PBASIC comments conflict with the code markdown (the first two comments are part of the code), so I've also provided some links]*
    ' {$STAMP BS2}
    ' {$PBASIC 2.5}

    DO
      TOGGLE 0
      PAUSE 500
      DEBUG "Toggling", CR
    LOOP

You can get the code from here:
    https://drive.google.com/file/d/0B0tev9Y3WaL4bGVmeGtsZ1lzb0E/view?usp=sharing

And you can see a little video demonstration here:
    https://drive.google.com/file/d/0B0tev9Y3WaL4WGZNQ3QwU0liOXc/view?usp=sharing

closed 2015-02-23 by PropGit

phated, 2015-02-23:
I updated the markdown to hopefully get everything contained.  How does that look?  Thanks!

PropGit, 2015-02-24:
Perfect!  Now I realize I was entering apostrophes instead of back ticks.  Thanks!

jacobrosenthal, 2015-02-24:
Note: I ended up using the documented
```
DEBUG "HI", CR
STOP
```