parallaxinc/Parallax-IDE #11: Research porting parallax programming protocol v1.1

jacobrosenthal, 2015-02-24:
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0CB8QFjAA&url=http%3A%2F%2Fforums.parallax.com%2Fattachment.php%3Fattachmentid%3D56837&ei=UMLrVI_NEJGRsQTEn4GoAw&usg=AFQjCNHViVe1WI4rt2cp387GBq-GYgfQuQ&sig2=CQQYHhQDCtUy3_b9hqWIlw&bvm=bv.86475890,d.cWc

jacobrosenthal, 2015-02-24:
Bare prototype at
https://github.com/jacobrosenthal/bs2-programmer/

PropGit, 2015-02-25:
Jacob,

I took a brief look at what you've done so far.  It looks like you've made
fantastic progress!

Respectfully,

Jeff Martin
Sr. Software Engineer, Parallax Incorporated
(916) 625-3002
jmartin@parallax.com
http://www.parallax.com

jacobrosenthal, 2015-03-06:
[bs2-programmer](https://github.com/jacobrosenthal/bs2-programmer) has been released as [0.1.0](https://github.com/jacobrosenthal/bs2-programmer/releases/tag/v0.1.0) and published to npm as [bs2-programmer](https://www.npmjs.com/package/bs2-programmer)

We demod in in the browser in an older branch of this app https://github.com/parallaxinc/ChromeIDE/tree/bs2-programmer

This sprint didn't require any graphical connection to programming so it simply run from the background page console with the command
```
window.bs2.upload('/dev/tty.usbserial-A502BMUQ', console.log.bind(console))
```

closed 2015-03-06 by jacobrosenthal