parallaxinc/Parallax-IDE #12: perform reset sequence

jacobrosenthal, 2015-02-24:


jacobrosenthal, 2015-02-24:
Started in on programmer document today and Ive got blocker on reset.

You guys use assert the Break condition. 
http://man7.org/linux/man-pages/man4/tty_ioctl.4.html

I've patched it into node-serialport, the native node serial library:
https://github.com/jacobrosenthal/node-serialport#break

And have a prototype uploader ready
https://github.com/jacobrosenthal/bs2-programmer/

But I can't find this functionality in the Chrome api. I've submitted a ticket to discuss with google: https://code.google.com/p/chromium/issues/detail?id=461169&q=break&colspec=ID%20Pri%20M%20Week%20ReleaseBlock%20Cr%20Status%20Owner%20Summary%20OS%20Modified

Some options I can think of to unblock:
* Is there an alternate upload method than using break?
* How was the TCP uploader solving this? Is that relevant here?
* Is there a firmware patch we can do for an alternate method?
* Is there a hardware modification we can do rather than support break? (ie something over  usb like avr does http://dicks.home.xs4all.nl/avr/usbtiny/)
* Does Propeller have same requirement? We could switch to propeller for prototype while we discuss with google.

@PropGit I've communicated with Iced Dev and they're aware. Lets get a call going when you're all ready.

PropGit, 2015-02-25:
> But I can't find this functionality in the Chrome api.
That's upsetting.  It seems both Apple and Google don't seem to realize they're missing a rare, but important, functionality in their serial APIs / drivers.
> Is there an alternate upload method than using break?
Not as far as the BASIC Stamp is concerned; upon reset (and once the Stamp's interpreter actually wakes up), it's SIN pin (Tx from host) must be asserted for longer than 9 bit periods at 9600 baud for it to realize that a host is trying to talk to it.  Otherwise, it simply boots up with the user's previously-downloaded program, as is normal.
> How was the TCP uploader solving this? Is that relevant here?
[We call it "downloading" actually (from host down to microcontroller); though it's debatable what is more correct, the term came from the fact that the host is often a way more powerful device than the microcontroller, similar to how web servers are often more powerful hosts than the clients that access them are.]  The IP-based (UDP) downloader has not been written for the BASIC Stamp yet, we designed it for the Propeller because the timing required by it is the harder of the two.  
> Is there a firmware patch we can do for an alternate method?
Not likely and very undesirable.  The interpreter is nearly full, I don't think there'd be room to implement an alternative unless it was very simple... but still undesirable because there have been over 1 million modules produced with firmware that expects this behavior.
> Is there a hardware modification we can do rather than support break? (ie something over usb like avr does http://dicks.home.xs4all.nl/avr/usbtiny/)
I just skimmed that page (hope I didn't miss the point), but I think I have a similar solution to propose that I'll get to in a moment.
> Does Propeller have same requirement? We could switch to propeller for prototype while we discuss with google.
No, but the Propeller has a timing window in which communication must start after reset; that can be tricky also.

Possible solution:
At 9600 baud, the byte period is approximately 1.04 ms.  We don't know exactly when the BASIC Stamp will wake up, but it should be about 18 ms after the end of the reset pulse... but circuitry and OS preempting can play a role in skewing what we "think" is the time it will wake up.  Regardless, once it is awake, the Tx line (connected to it's SIN pin) *must* be asserted (not resting) for longer than 9 bits (or more likely, > 10 bits; > a byte period), so it must be sufficiently long for the BASIC Stamp to interpret what it sees on the Tx line as a break condition.

So, ideally, it would be best if Google implements break support in the API; however, until then, we can probably get what we need by setting the baud rate to something very low and transmitting a single $00 (which should result in a start bit (asserted) followed by eight 0 bits (looks asserted) and a stop bit (resting) that will appear to be a short break condition.  150 baud would be great, but the slowest the FTDI USB-to-Serial chips (included on our boards) can go is 183.1 baud... which gives us an "interpreted" break condition period of 49.152 ms (or > 47 byte periods of 9600 baud data, and if the reset pulse is 2 to 10 ms in length, is greater than the recommended minimum 36 ms of break condition after reset)... so:
1. Before reset, set the baud rate to 183 (may have to be even, if I recall, so perhaps 184)
2. Generate reset pulse
3. Transmit $00
4. Change baud rate to 9600
5. Transmit data...
The key is ensuring that the baud rate doesn't really change until after the byte in the buffer is fully transmitted (I'm assuming that will happen naturally, but we may be best off waiting until notified that the Tx buffer is officially empty before performing step 4.  Does the API have such TX Buffer empty notification functionality?).

One odd thing I noticed a while back that might come into play here is that the DTR toggle and serial transmission may occur in a different order than you specified.  I think I had to code it backwards in order for it to happen in a consistent way; which is very strange.  Keep a look out for that kind of thing.

Do you have a logic analyzer so you can test the resulting signals?  If not, is there a way I could test it and show you the results?

> Lets get a call going when you're all ready.

Sure, Wednesday afternoon is open for me.

jacobrosenthal, 2015-02-25:
That MIGHT work. from chrome api:

The requested bitrate of the connection to be opened. For compatibility
with the widest range of hardware, this number should match one of
commonly-available bitrates, such as 110, 300, 1200, 2400, 4800, 9600,
14400, 19200, 38400, 57600, 115200. There is no guarantee, of course, that
the device connected to the serial port will support the requested bitrate,
even if the port itself supports that bitrate. 9600 will be passed by
default.
Ill investigate.

On Wed, Feb 25, 2015 at 12:12 AM, Parallax Git Administrator <
notifications@github.com> wrote:

> logic analyzer so you can test the resulting signals? If not, is there a
> way I could test it and show you the results?

PropGit, 2015-02-25:
The FTDI drivers will automatically take a requested baud rate and apply the closest possible baud rate that the chip can perform... in this case, requesting 110 would probably result in 183.1.  That's assuming that the computer is using the FTDI drivers, or some other driver that supports the same functionality for the FTDI chipset.  I think one of my coworkers determined that FTDI drivers were automatically included with Linux (and Chromium), but I'm not positive.

There are no more computers produced today (that we've seen) that have actual DB9 serial ports on them... leaving us just USB ports.  All BASIC Stamp development hardware produced by us uses the FTDI FT232RQ or FT231X chips for USB-to-Serial functionality (either built into the board itself, or provided via an adapter we sell)... so hardware compatibility can be safely assumed to be a small set of end devices.  The "potentially" bigger problem is whether or not the computer itself is using FTDI drivers.

On Windows, it automatically gets the latest FTDI drivers from Windows Update the moment you plug in an FTDI-based device.

On Linux (Ubuntu, at least) the FTDI drivers are already included with the OS (from what I've been told).

On Mac, Apple has "done it again"... starting with Mavericks, the began imposing their own kext for FTDI chips, which unfortunately does not have support for break (as far as I know) and no support for actually setting and clearing DTR manually.  You can still install the FTDI driver on the system, but it's been known to cause kernel panics in some cases... we presume (but haven't proved) that it's because they also changed the prescribed location where the kext should be stored.

On Chromebook... well, I don't really know the situation there yet, regarding FTDI drivers.  Do you know?

jacobrosenthal, 2015-02-26:
*That's upsetting. It seems both Apple and Google don't seem to realize they're missing a rare, but important, functionality in their serial APIs / drivers.*
Please tell them that and star our issue :) We did get some action there today so that might be a good sign:
https://code.google.com/p/chromium/issues/detail?id=461169&q=break&colspec=ID%20Pri%20M%20Week%20ReleaseBlock%20Cr%20Status%20Owner%20Summary%20OS%20Modified

*The key is ensuring that the baud rate doesn't really change until after the byte in the buffer is fully transmitted (I'm assuming that will happen naturally, but we may be best off waiting until notified that the Tx buffer is officially empty before performing step 4. Does the API have such TX Buffer empty notification functionality?).*
Yep, theres a write complete callback in chrome's api.

*On Chromebook... well, I don't really know the situation there yet, regarding FTDI drivers. Do you know?*
We've done serial on Chromebooks, but I dont know which driver they use. I can look at that next once we have something up in Chrome in general.

@phated brought in the build system today so took a look at some signals.
Chrome errored on most (all?) non standard bauds I tried including 120, 130, 140, 160, 170, 180, 190. 110 and 150 didnt error, but didnt seem to put out waveforms. 200 was successful and a 0x00 write generated a 45ms signal. My existing sketch is actually resetting with only a 20ms so this seems more than fine. 

So far I havent had success with chrome's update api to swap to a new baud, it returns success condition but my next write appears to be at the same baud.

I did notice that the dtr reset was unnecessary in my current code. I can see a very brief blip being generated I presume by the port being opened, a 'feature' Ive used before for programming arduinos.

Im also working on just closing and reopening the port at the new baud--but where I left it that was acting poorly. I will say Im worried about how the dtr toggling will affect that.

jacobrosenthal, 2015-02-27:
I have a test branch up that opens a port at 200, sends a byte, changes the baud to 9600 and sends a byte-- which doesnt work
https://github.com/parallaxinc/ChromeIDE/tree/serialtest

It actually does work with certain other bauds and directions, like 9600 -> 200 and 4800 -> 9600 .. 

Before I submit this (new) chrome bug Id like to rule out the ftdi chip
@PropGit Do you guys have a db9 machine lying around still? I know until recently you still had db9 bs2's around in the wild.

jacobrosenthal, 2015-02-27:
Oh and it needs to run Chrome latest obviously :)

jacobrosenthal, 2015-02-27:
OK another test, which might make it more likely FTDI's problem not Chrome's
https://github.com/parallaxinc/ChromeIDE/tree/serialtest-open-close

an open, close, open exibits the same characteristics
200   -> 9600 maintains 9600
9600 -> 200 works as expected

PropGit, 2015-02-28:
> Please tell them that and star our issue :) 

Okay, I did.

> 110 and 150 didnt error, but didnt seem to put out waveforms.

You mean, it didn't generate any serial output at all?

> 200 was successful and a 0x00 write generated a 45ms signal. My existing sketch is actually breaking with only a 20ms so this seems more than fine.

The BASIC Stamp will typically reset and wake up in 18 ms after DTR deasserts.  We recommend 2x 18 ms (or 36 ms after DTR deassertion) but we'll take what we can get.  The unknown is how consistently-timed the serial stream will be with the edges of DTR.

> So far I havent had success with chrome's update api to swap to a new baud, it returns success condition but my next write appears to be at the same baud.

Uh oh.

> I can see a very brief blip being generated I presume by the port being opened...

Yes, port opening tends to assert DTR from what I remember, then our code deasserts it as fast as possible.  Not much we can do about that.

> Im also working on just closing and reopening the port at the new baud--but where I left it that was acting poorly. I will say Im worried about how the dtr toggling will affect that.

That will re-reset the BASIC Stamp, likely nullifying the whole effort.

>It actually does work with certain other bauds and directions, like 9600 -> 200 and 4800 -> 9600 ..
>Before I submit this (new) chrome bug Id like to rule out the ftdi chip
>@PropGit Do you guys have a db9 machine lying around still? I know until recently you still had db9 bs2's around in the wild.

I'll do some tests this weekend.  I don't think I have a system with a DB9 handy, but we may still have a machine in Manufacturing I could do some testing with.  I have many other obligations for the next 4 days... I'm hoping I can squeeze some significant time into this though.

I'll update when I've tested.  After my own tests, I'll probably want to test your branch.  I'm assuming the instructions in the repo will help me get it running?

PropGit, 2015-03-01:
Okay, my first tests were to resurrect an old serial port test program I wrote for Windows.  I modified it to do the same kind of this you're trying and I watched the signals with my logic analyzer.  I'm using my normal development system, with USB ports and our FTDI FT232RQ-based USB-to-Serial adapter.

I have two good news items to report... 
  1. I can duplicate most of what you're describing with my native Windows code (on a system with the FTDI driver installed).
  2. I see what's going on now.

110, 120, 130, 140, 150, 160, 170, 180 don't cause errors in my test, but they result in an effective baud rate of 115.2 kBaud, strangely.

184 and 190 work and generate exactly the prescribed baud rate.

200 works perfectly as well.

Switching from 200 to 9600 *seems* to generate only 9600 waveforms, while switching from 9600 to 200 generates the expected signals.

After extensive tests, I got lucky and happened to notice exactly what's going on... The system/UART *is* accepting the request to change the baud rate from 200 to 9600 (or 9600 to 200) and does so even if data is still being transmitted.  By "lucky," I mean that when I was transmitting from 200 to 9600, I happened to get a trace where the first couple bits of my first byte (I was using $55 at the time, for both bytes) were transmitted at 200 baud, but then suddenly there was a short bit followed by bits that were exactly timed at 9600.  It changed baud rates mid-bit!  But most of the time, it changed to 9600 before even the first bit of the first byte was transmitted; which is what you were seeing.  The reason it didn't happen from 9600 to 200 is likely that it just took so little time to transmit the first byte in that case.

Based on this, I don't think there's anything wrong with the Chrome serial API, other than it doesn't support break conditions yet.

We definitely need to ensure that the first byte is finished transmitting before switching to 9600 baud.  

Have you already coded the write complete callback in your code?  In my attempts to check for Tx empty in my code, it always reports empty, so the results are no different (unpredictable).  I think I'd have to do an overlapped write operation to get the results I need using that mechanism.

However, I reasoned that I could get better results right now by timing it better; and I have.  Using the proposed 200 baud to 9600 baud switch technique, and with a little experimentation and measurement, I've been able to get pretty consistent results where the reset pulse stays near the start of the simulated break condition (sometimes before and sometimes after), the "break" lasts for 45 ms, and the baud rate switches consistently after the initial $00 is sent before the next byte at 9600 is sent.

Here's a captured sample.  The long Tx low is a $00 byte transmitted at 200 baud, followed by a 75+ ms delay, followed by a $55 transmitted at 9600 baud.  The reset pulse occurs (in this example) about 5 ms after the first falling edge of Tx.

[image]

If I recall, we have about 2.3 seconds after the end of the "break" condition before the BASIC Stamp gives up (if it doesn't receive the first byte of the host's serial communication).  In this case, I simply delayed for 100 ms immediately after the call to transmit the first byte, before switching baud rate to 9600 and transmitting a $55 (for example sake).

As I alluded to in an earlier message, the relative signal instances (DTR and Tx) and resulting consistency seems to work better slightly out-of-order.  Here's the algorithm I found to work well for this test:

  * Open the port at 200 baud
  * Transmit $00
  * Assert DTR
  * Delay 2 ms
  * De-assert DTR
  * Delay 100 ms
  * Set port to 9600 baud
  * Transmit $55
  * Close port

Note: The "Delay 2 ms" is intended to prevent the assert/de-assert from happening so fast that it ends up basically nullified by the OS.  The "Delay 100 ms" was to allow enough time to transmit the first byte, assuming the OS delivered the byte to the UART soon enough.  The final could be a fixed delay (somewhere in the 100 ms to 1000 ms area would probably be best).  The $55 is just for testing; the point is to send the actual Identification bytes at that point, instead.

Also, I never experienced DTR asserting upon port opening in my tests tonight.  What are you using as a development box?  OS X?  If so, one of our developers noticed strange behavior is OS X, like DTR simply toggling instead of asserting or de-asserting on demand.  I'd like to know if you experience this.

jacobrosenthal, 2015-03-02:
@PropGit Looks like you were right. I assumed our write callback was upon completion, but by adding >70ms blocking delay after the write before I update the baud settings I have it working now:
https://github.com/parallaxinc/ChromeIDE/tree/bs2-programmer

jacobrosenthal, 2015-03-06:
I've patched node-serialport upstream to support reset via the BRK control signal which landed tonight as  [node-serialport 1.6.1](https://github.com/voodootikigod/node-serialport/releases/tag/v1.6.1). 

In chrome we use a wrapper library called browser-serialport which I actually maintain.  There we're using the fake break via a baud update. Ironically we haven't landed that yet official yet, but work is going off of the [update branch](https://github.com/garrows/browser-serialport/tree/update)

But we should still keep pushing on our Chrome ticket long term.

closed 2015-03-06 by jacobrosenthal

PropGit, 2015-04-09:
+jacobrosenthal Did you see that someone replied with code to the Google issue you submitted?

Comment #6 on issue 461169 by clivepau...@gmail.com: Serial api doesnt support sending Break
https://code.google.com/p/chromium/issues/detail?id=461169

If someone would like to have a go at adding this essential RS-232 standard feature as something like

chrome.serial.setBreak(integer connectionId, boolean breaked, function callback)

the important 'break specific' bits would be coded as follows:

```
serial_connection.h

virtual bool setBreak(bool breaked);

serial_connection_posix.cc

bool SerialConnection::setBreak(bool breaked) {
        if (breaked)
                return ioctl(file_, TIOCSBRK, 0) == 0;
        else
                return ioctl(file_, TIOCCBRK, 0) == 0;
}

serial_connection_win.cc

bool SerialConnection::setBreak(bool breaked) {
        if (breaked) {
                if (!SetCommBreak(file_))
                        return false;
        } else {
                if (!ClearCommBreak(file_))
                        return false;
        }
        return true;
}
```

Hope that helps towards getting this issue fixed!

jacobrosenthal, 2015-04-09:
I did! Hoping for a google reply so we know where their head is on the issue.