parallaxinc/Parallax-IDE #29: Tokenizing

phated, 2015-03-09:
Need to research and implement tokenizing of PBASIC code.

Options include:
* Use emscripten to compile tokenizer to JS
* Rewrite the tokenizer in JS 
* Send files to a server that tokenizes using C tokenizer

phated, 2015-03-10:
@PropGit can you include David in on this thread?  I'd like to see his steps and results related to emscripten.

PropGit, 2015-03-11:
I prompted David to jump in here.

I should note that there's a fourth possible option that David mentioned; A JavaScript-based x86 emulator/VM.

In an earlier conversation, David said:
There is a very simple open-source implementation here: https://github.com/copy/v86.  

Also, someone wrote a JavaScript program to interpret the assembly mnemonics: http://timhatch.com/projects/jsassembly/index.html and it didn't take much.  (Click view source after opening that last link.)

As far as I know the JavaScript VM is interpreting at least a modern 32-bit instruction set.  It is really fast too, it only takes 60 seconds to boot Linux: http://bellard.org/jslinux/.  (That implementation uses an i586 Linux kernel).

The last link is a closed-source project unrelated to the Github implimentation I linked to earlier.  The one on Github has a bootable Linux system here: http://copy.sh/v86/?profile=linux26 and several other operating systems here: http://copy.sh/v86/.  It looks like at least two people have written a Linux-booting x86 VM in JavaScript.  The closed source project has a page about it: http://bellard.org/jslinux/tech.html that mentions that "the x86 emulator can be modified to provide an API to load x86 dynamic libraries and to provide a js-ctypes like API to call the C/C++ functions from javascript."  Also, on the closed-source implementation, try running "time tcc -o hello hello.c" then "./hello".

dcarrier-parallax, 2015-03-11:
For some background on Emscripten:

Most multi-language compilers, like GCC and LLVM, compile from a given language to an intermediary language then from the intermediary language to the target architecture.  Emscripten compiles the input code using the LLVM compiler, but instead of providing a target architecture, it grabs the compiled intermediary language, (also called LLVM) and wraps a JavaScript interpreter around it.

The performance and complexity would be on par with using one of the aforementioned x86 interpreter VMs with our existing x86 binaries.  There is more much support available for Emscripten than the x86 interpreters, so it would likely be the easier route of the VM options.

In neither case can you modify the code once it has been compiled.  Performance probably won't be a problem, but a native JavaScript translation of the code will likely be several times faster.

My preferences between our various options are as follows:

* Translate the tokenizer code from C to JavaScript
* Use an Emscripten VM to interpret the existing C code
* Use an x86 interpreter with existing binaries
* Send the files to a server for remote compilation

I would prefer we not go with the last option, as it would take a significant investment on our part to create the remote compile system and to keep it secure, it could be a significant liability if anything goes wrong on our end, it will require constant network uptime for us and the user, it creates a poor user experience, and it can be difficult to use behind a firewall.

— David Carrier
Parallax Inc.

jacobrosenthal, 2015-03-16:
TLDR we have a working but not perfect prototype but were blocked on https://github.com/parallaxinc/ChromeIDE/issues/34

Last week wewere able to run emscripten against the tokenizer.cpp file and generate javascript. Full details in this gist https://gist.github.com/jacobrosenthal/92889db5223891a985a8

However there appears to be no official support for passing a struct back. I'm faking it by passing a big uint8 array in but the memory locations are a little odd on the return trip. Probably because of struct packing http://www.catb.org/esr/structure-packing/ 
This makes me think my solution wouldnt build consistently on 100% of architectures -- which is probably why theres no official support. That said, nobody else NEEDS to build it once you release it.

@phated has work he can publish on how to add the stampapp example to the build. We could use that to pull out primitive data structures like just the resulting code, and safely pass that every time. @PropGit perhaps we can discuss whats needed or have your team decide on a conversion of that struct to a primitive type like an array or array of arrays?

We noticed another minor problem in the Version command in javascript. I was working off version 123 which is stored as a uint8t and returned properly to javascript, @phated was working on 130 which returned as -126, presumably rolling over. Maybe needs to be filed with emscripten people?

Also @PropGit is there a struct member for size of the eeprom block? When programming Im currently just pulling out all 2048 bytes from memory but bs2 is unhappy to be sent 0s.

phated, 2015-03-17:
Doing some more digging today, we found the figure on page 7 of the "Using the PBASIC Tokenizer Library v1.2.pdf" that outlines the struct member alignment.  Based on this, we see 5 bytes of spacing in the first 11 bytes of the memory structure.  We've come up with the following JavaScript code to demonstrate the spacing is correct:

```js
var buf = new Buffer(result);
console.log('succeeded', buf[0]);
console.log('error', Module.Pointer_stringify(buf[4] + (buf[5]<<8) + (buf[6]<<16) + (buf[7]<<24)));
console.log('DebugFlag', buf[8]);
console.log('TargetModule', buf[9]);
console.log('TargetStart', buf[12] + (buf[13]<<8) + (buf[14]<<16) + (buf[15]<<24));
```

Given this and the fact that the struct elements are 6500 bytes but the full struct is 6508 bytes, we believe the last 3 bytes of spacing are after PacketCount (a one byte field).  Is this correct @PropGit??  If so, I believe we have enough information to change the goal from server compiling to in-browser compilation. Let's have a call about this tomorrow.

PropGit, 2015-03-18:
> I'm faking it by passing a big uint8 array in but the memory locations are a little odd on the return trip.

The tokenizer itself just needs the caller to pass it an address of a long-aligned (4-byte aligned) contiguous memory block (of proper size in bytes) that it can write to.  The tokenizer will write to it based on the designated structure element positions (as indicated in the document) and with byte padding in-between some structure elements as needed to ensure the next logical data element is aligned on a 4-byte boundary if, and only if, it's element(s) unit size is greater than 1.  As long as you can ensure the memory is long-aligned (multiple-of-4 address) and is the proper size (rounded up in bytes to fill a multiple of 4), it should work on every platform that it is compiled on.

> We noticed another minor problem in the Version command in javascript. I was working off version 123 which is stored as a uint8t and returned properly to javascript, @phated was working on 130 which returned as -126, presumably rolling over. Maybe needs to be filed with emscripten people?

This is concerning.  Seems like an emscripten problem.

> Also @PropGit is there a struct member for size of the eeprom block? When programming Im currently just pulling out all 2048 bytes from memory but bs2 is unhappy to be sent 0s.

The EEPROM block is only intended for generating an memory map (a visual for the user), in which case, all 2048 bytes are necessary always.  It's the PacketBuffer you're after.  The PacketCount field just before the PacketBuffer indicates the number of 18-byte packets to read from the PacketBuffer (starting at element 0).  Each 18-byte packet is exactly what needs to be sent to the BASIC Stamp; they include the EEPROM block number and checksum values as well as 16-bytes of data.

> Given this and the fact that the struct elements are 6500 bytes but the full struct is 6508 bytes, we believe the last 3 bytes of spacing are after PacketCount (a one byte field). Is this correct @PropGit??

Well, I think you're almost right.  Because VarCounts is long-aligned, PacketCount is long-aligned, but PacketBuffer is not long-aligned (because it's a byte array)... but, it's a multiple of 4 bytes in length, leaving it 3 bytes short of filling the last long of the buffer.  Thus, PacketBuffer element 0 appears immediately after the PacketCount byte, but after PacketBuffer element 2303, there are 3 more bytes needed to pad to the next long-aligned memory location.  So, if the memory you're allocating is based on a unit of 4 bytes in size, it would naturally end up 6508 bytes in length.  If the memory is allocated as units of bytes, it can be 6505 bytes in length; but you must ensure the buffer is somehow long-aligned so that you can access the character pointers and ints that make up parts of the structure.

> If so, I believe we have enough information to change the goal from server compiling to in-browser compilation. 

!!!!!! REALLY ?!?!?  That's awesome!  Using Emscripten-ized PBASIC Tokenizer?  Wow!  That's so fantastic!

> Let's have a call about this tomorrow.

Yes, let's have a call about it - unless I just gave you all the answers you need to proceed.  I think I'll be available until as late as 3 pm PT.

PropGit, 2015-03-18:
I forgot to say that there's a built-in function for testing the structure alignment (and field alignment), called TestRecAlignment.  It's mentioned on after the table on page 7, and described in more detail on pages 9 and 10.  You should call that function and then verify that the data returned in the buffer you delivered matches what's indicated in the table on page 10.

PropGit, 2015-03-18:
> We noticed another minor problem in the Version command in javascript. I was working off version 123 which is stored as a uint8t and returned properly to javascript, @phated was working on 130 which returned as -126, presumably rolling over. Maybe needs to be filed with emscripten people?

>This is concerning. Seems like an emscripten problem.

Wait... maybe not.  I looked at how Version is defined.  I wrote in the document that Version returns an int.  But in the code it's written as STDAPI which is defined in the header as uint8_t if being compiled on Linux.  I don't know why it's defined like this (and can't remember if I did it or if a developer who was helping me with porting issues did it... but may have been me)... now I think it should be a 32-bit unsigned integer value.

Maybe the problem isn't really emscripten?

jacobrosenthal, 2015-03-20:
>It's the PacketBuffer you're after

You're right, I just misnamed it because I wasnt sure of the memory mapping at first.

>Given this and the fact that the struct elements are 6500 bytes but the full struct is 6508 bytes, we believe the last 3 bytes of spacing are after PacketCount (a one byte field). Is this correct @PropGit??

Now that weve mapped it all up its looking like the last 3 bytes of padding are actually AFTER PacketBuffer. The tests should confirm this.

If you like, you can take a look at the branch Im working on in https://github.com/parallaxinc/PBASIC-Tokenizer/tree/emscripten

>I forgot to say that there's a built-in function for testing the structure alignment

My current feeling is theres isn't a use case for writing anything around that function atm as we need to write tests to catch any errors in our code and that should catch anything at that level as well.

>Maybe the problem isn't really emscripten?

Seems like youd probably like it to be bigger than uint8_t so you have some version headroom, but uint8_t still properly holds up to 255 so it seems like emscripten.

jacobrosenthal, 2015-03-20:
@PropGit TestRecAlignment turned out to be a very nice test actually. I THINK the only thing missing currently is the Source-to-Token-Reference Structure and the version problem. Ive added them both as issues to that repo to be dealt with time permitting.

phated, 2015-03-24:
This has been merged to master

closed 2015-03-24 by phated