parallaxinc/Parallax-IDE #30: Implement a release process.

phated, 2015-03-09:
As stated by @PropGit today, there will probably be a beta release at the end of this sprint, so we want a release process implemented.

phated, 2015-03-09:
Chrome webstore has a REST API - https://developer.chrome.com/webstore/using_webstore_api

I think it would be really cool if Travis published a release to "GitHub Releases" and then to the Chrome Webstore upon creation of a new tag.

phated, 2015-03-10:
I've verified that the REST API will work for this process.  The application will need to be created in the Chrome Webstore first, as it needs a description, images, etc and then we can add the ItemID to the release process.

phated, 2015-03-10:
Info on using GitHub Releases with Travis CI - http://docs.travis-ci.com/user/deployment/releases/

PropGit, 2015-03-17:
Is it still possible to do this while keeping the application for beta test and "private" on the Chrome Webstore?

phated, 2015-03-17:
yes

phated, 2015-03-24:
We have this working in travis.  There's some stuff I need to do with @PropGit to add it to the Chrome Webstore and then we can discuss automating the Webstore update

closed 2015-03-24 by phated