parallaxinc/Parallax-IDE #33: Getting a timeout on bootload

jacobrosenthal, 2015-03-10:
Every randomly we get a timeout when bootloading from inside chrome. So far its been non-reproducible with the same code in node osx. I assume its in chrome serial or browser-serialport (our chrome serial wrapper) layer. 

To start I recommend re-scoping the two to see if theres an obvious timing difference

```
app.bs2serial.bootload({path:path}, function(error, result){console.log(error,result)})
Promise {_handler: Pending, then: function, catch: function, _beget: function, spread: function…}
VM170:2 Error: Sending 42: receiveData timeout after 1000ms {stack: (...), message: "Sending 42: receiveData timeout after 1000ms"} undefined
```

jacobrosenthal, 2015-03-11:
The only difference I could find was an extra 2 ms delay (15 to 17ms) between the reset signal and the beginning of the break signal on a 'bad' bootload. I actually cant control that slice of time so I tried @PropGit 's suggestion in #12 of sending the break byte first and so far I haven't seen a bad bootload yet. 
Fixed in dep https://github.com/jacobrosenthal/bs2-serial/commit/a7061227f62d44d840d76b06d0a36689af74d668 but leaving open for the week for more results.

PropGit, 2015-03-11:
Great!  Hopefully the break-immediately-before-reset method works reliably.  I've certainly seen good evidence to support that.  

Make sure there's no "coded" delay between the two, only after the reset; it's the best we can hope for unless/until Google supports Break in their serial API.

phated, 2015-03-19:
None of us have seen failures since @jacobrosenthal's change.  Closing.

closed 2015-03-19 by phated

PropGit, 2015-03-19:
That's fantastic!