parallaxinc/Parallax-IDE #48: Display pane for output unformated serial console data in the UI.

phated, 2015-03-25:
Integrated into the IDE window, probably below the editor.  Always open unless opening/closing works better for the architecture.

PropGit, 2015-03-25:
Here's some implementation notes.  Items *not* required for proof-of-concept are marked __FUTURE__, and are included here only as a "heads up" about later needs.

See issue #47 for related information.

The process is:
* Upon data arrival, output data to display in a left-to-right, top-to-bottom scrolling fashion
  * Data may contain control characters that cause special display behavior
    * ASCII 32 to 127 - display as-is
    * ASCII 13 - perform new-line operation
    * ASCII 0 to 12, 14 to 31 and 128 to 255 - ignore
    * __FUTURE:__ ASCII 0 to 15 perform special functions like Clear Screen, Cursor positioning, clearing, new lines, and sounding of a "chime" sound

phated, 2015-04-09:
@PropGit I am not seeing the control characters coming across with the content from the `debug` statement.  I have an open question to @jacobrosenthal related to this, but any insight would be helpful.

PropGit, 2015-04-09:
Maybe there's a misconception here... does the DEBUG statement have control characters listed in the code?  It doesn't output control characters unless told to.

For example:
```  DEBUG  "This is a test", CR
will output an ASCII 13 character after the "This is a test" text.

```DEBUG "some text", 0

will output an ASCII 0 character after the "some text" text.

phated, 2015-04-09:
Thanks.  We got it figured out awhile ago.  Sorry for not updating the thread.

PropGit, 2015-04-09:
No problem.  I've been busy with other projects for the last 5 hours or so and just didn't get to this until now.

PR #63 (2015-04-10): communicate with BS2 over serial, add output pane for data, clear data on download - closes #47, closes #48, closes #49
#### What's this PR do?
Adds support for BS2 communication over serial and displays the data in an output pane positioned under the editor.  It also clears the output pane before downloading to the BS2.

#### Where should the reviewer start?
Pull down the application and rebuild.

#### How should this be manually tested?
I've been using the following code snippet to test:
```
'{$STAMP BS2}
'{$PBASIC 2.5}

Counter VAR BYTE 'Counter for loop
FOR Counter = 1 to 20 'Loop 20 times [... cut]
commit 389fd58b (2015-04-10): communicate with BS2 over serial, add output pane for data, clear data on download - closes #47, closes #48, closes #49
commit ed10673f (2015-04-10): Merge pull request #63 from parallaxinc/serial-output-data

communicate with BS2 over serial, add output pane for data, clear data on download - closes #47, closes #48, closes #49
closed 2015-04-10 by monteslu