parallaxinc/Parallax-IDE #51: Display file saved/unsaved state next to filename in sidebar listing.

phated, 2015-03-25:
This could be represented by green and red dots next to filenames.

PropGit, 2015-03-26:
After some thought, I think it may be best to use
* an icon of a pencil writing something (red, perhaps) when the file has been edited but unsaved by user
* no icon (just blank) when the file has not been edited (or was saved by the user)

phated, 2015-04-08:
@PropGit I tried getting your suggestions to work, but couldn't easily within the current implementation of the sidebar and I didn't think it was pertinent to completely redo it.

PropGit, 2015-04-08:
Okay, let's focus on other issues at this point.

phated, 2015-04-08:
To clarify, I've implemented as originally noted in the issue because it was straightforward to add, but the icons, etc from the suggestion were harder to get right.

PropGit, 2015-04-08:
Thanks!  I was wondering that and hoping that was the case.

PR #58 (2015-04-08): Add temp file support
#### What's this PR do?
Adds support for temp files and indicates in the UI when a file has unsaved changes (a temp file).

#### Where should the reviewer start?
Pull down these changes and rebuild the application.

#### How should this be manually tested?
Create a file in the application, add some text, save it, then add some more text, the indicator should change between green and red.

#### Any background context you want to provide?
Nope

#### What are the relevant tickets?
Closes #51
Closes #52 

#### Screenshots (if appropriate)
[image]
[image]
closed 2015-04-08 by monteslu