parallaxinc/Parallax-IDE #53: Create icons for app

PropGit, 2015-03-26:
Create icons for editor project "ChromeIDE." 
* 16 x 16
* 48 x 48
* 128 x 128

.png format.

PropGit, 2015-03-26:
Can you show me examples of where these icons would appear?  16 x 16 is very constraining; hard to convey much there.  Need to know context.

Also, is it full 32-bit color (24-bit RGB + 8-bit alpha)?

phated, 2015-03-26:
I believe places the icons appear are outlined at https://developer.chrome.com/apps/manifest/icons

Yes, full 32-bit color is correct

PropGit, 2015-03-26:
Thanks, that helps a lot.

PropGit, 2015-04-02:
We've created many icon (ideas) for this app and have settled on these for now.  Can you include these in the repo and in the next builds?

[image]

[image]

[image]

commit 79ed04be (2015-04-05): add icons to the application - closes #53
closed 2015-04-05 by phated