parallaxinc/Parallax-IDE #54: Spot-test Emscripten-ized PBASIC Tokenizer

PropGit, 2015-03-27:
We forgot this one... for Sprint 3, we need to at least spot-check the PBASIC Tokenizer that is Emscripten-ized.  By spot-check, I mean try a few random pre-defined tests, including at least one large PBASIC program, each from the test suite.

If *__all__* the tests can be performed in Sprint 3, that'd be great, but if time it too tight, let's stick to a small number of tests.  

Agreed?

Here's a video I made for you describing [how the existing test interface works](https://drive.google.com/file/d/0B0tev9Y3WaL4QUYtN3JHR2o4Sm8/view?usp=sharing).

The test cases are in the Tests.txt file located in the archive I sent you previously, in the "Linux x86" subfolder.  Also in there is the testtokenizer.cpp and testtokenizertypes.h and makefile that I used to make a wrapper around the tokenizer that pumps the test cases through the Linux or Mac-based tokenizer.  It would probably still work today nearly as-is... maybe a couple minor changes are necessary for today's GCC.  Regardless, you can look at the testtokenizer code to see how you could form a JavaScript-based test wrapper around the Emscripten-ized PBASIC Tokenizer and pump all these tests through it to generate a Results.dat file.  With that resulting .dat file, you (or I) could import it into the existing Windows-based testing interface to see if all the tests passed or failed in any way.

NOTE: in the video I show the full suite of test cases being 1,600+, but the Tests.txt file that you have is a little shorter (about 1,300 tests if I remember right) because it's intentionally missing the BS2px tests.

phated, 2015-03-27:
This may have to fall in Sprint 4 since we didn't discuss it as a task during the planning meeting.  Let me discuss with @jacobrosenthal

phated, 2015-04-08:
Given the deadline for Sprint 3 is approaching, I'm going to remove this from the sprint.  Also, I am going to mark it as an enhancement because all it will really be doing is being a smoketest that Emscripten is working correctly, which, I believe, is already heavily tested.

PropGit, 2015-04-08:
Agreed on Enhancement label and removing from Sprint 3.  We'll need to address it in a later sprint to ensure our overall solution is sound.

PropGit, 2015-05-13:
__[From 2fast2fourier]__

I've started on [this issue], and I've placed an initial implementation of this test suite in the 'testsuite' branch of the PBASIC-Tokenizer repo. It depends on the SourceSize PR, so it won't run on a fresh build until that is merged.

Right now execute every test case in the test file. For each test it will compile the source, and verify that 'Succeeded' matches the PASS/FAIL flags, and that 'Error' matches the FAIL condition message. So far only 1 test in 1365 cases is failing, but until the remaining checks are implemented we might have false positives.

The test case thats failing is testing strange character support:
https://gist.github.com/2fast2fourier/e3e64d63d9bbaa9d79d1

That build should pass, but the actual failure message I receive is:

ErrorStart: 5516,
ErrorLength: 8,
Error: '111-Undefined label'

Can you look at the example and let me know if there is anything that might cause the '111-Undefined label' failure based on syntax alone?

I also tried to run the example in MacBS2 3.1.8(prerelease) and I could not compile until I removed the non-ascii comments. It gave an `Unknown target module. $STAMP directive not found` error in the UI, I verified I only had one '{$STAMP BS2} entry.

PropGit, 2015-05-13:
That's fantastic progress!  I understand the high potential for false positives.

Okay, that's test number 815.

The error you're getting doesn't make sense in terms of just pure code syntax because that ErrorStart location it reports (5516) is right in the middle of the LOOKUP command (keyword).

Those "strange characters" are part of the ASCII character set, but they are in the Latin-1 extension (128 - 255) so what I suspect is happening is that the source you're feeding to the tokenizer is UTF-8 encoded, in which case all the normal characters the tokenizer expects happens to be encoded as single bytes 0-127 (perfect), and the Latin-1 extension characters are encoded at two-byte values (uh oh) that relate to the Unicode code points 128-255.

The tokenizer was not written to handle anything but single-byte characters of 0-255.  

Going forward, the source can be made to use UTF-8 encoding, or UTF-16, or whatever, with international characters in comments too, but before sending to the tokenizer, it all needs to be converted to ANSI encoding with any characters that are outside of the 0-255 range translated to something proper... which I'll have to point out what "proper" is later (something that will cause a proper syntax error where need, and will work in comments just fine too.

It's interesting that MacBS2 has a similar problem.  I guess the developer didn't test those strange characters.

phated, 2015-05-15:
The one outstanding test case that has problems has been documented at https://github.com/parallaxinc/PBASIC-Tokenizer/issues/7  - closing this issue.

closed 2015-05-15 by phated

jeffam70, 2015-05-15:
To follow-up on my previous post:

Regardless of source code encoding employed in file/database/internal, the source that is passed to the tokenizer must be ANSI encoded (single-byte characters) where:
  * Any single or multi-byte character that represents a specific ASCII character (0-255) is converted to a single-byte 0-255 value, and,
  * Any multi-byte character that represents a non-ASCII character (> 255) is converted to the single-byte semicolon character ';' (decimal 59, hexadecimal 3B).

The above should _only_ happen in source delivered to the tokenizer, __not__ source visible to the user in any way.

NOTE: For all cases, Tokenizer's Start and Length responses refer to character counts, not byte counts, although the two interpretations may be equivalent in certain encodings.

ALSO NOTE: The semicolon character was chosen as a proper representation of a non-ASCII character for tokenizing purposes because
  1) It is not used by the PBASIC language, nor the Spin or PASM languages, to represent anything
  2) It causes a very fitting "Unrecognized character" error message.