parallaxinc/Parallax-IDE #56: Limits on Browser Local Storage and Temporary Storage

PropGit, 2015-04-06:
Browser Local and Temporary Storage is of huge importance to this project.  I like this feature and our use of it so far.

Are there limits imposed on Local Storage from the Chrome Browser?  I just read an article from 2013 that said it's something like 4 MB.

How about Temporary Storage?

Is either Local or Temporary Storage automatically deleted by the browser at any time, or by some user action?

phated, 2015-04-06:
Chrome applications are allowed 5mb of local storage by default but they can request `unlimitedStorage` as a permission in the manifest to get an unrestricted storage amount.

More information: https://developer.chrome.com/extensions/declare_permissions

phated, 2015-04-08:
@PropGit if we add `unlimitedStorage` to the permissions. Can we close this?

PropGit, 2015-04-08:
Yes, that seems like the best solution.  Awesome. You can close this issue after adding unlimitedStorage.  Is that done in the manifest? 

I should point out though that I expect the file manager code you build (in future revs, not now) to gracefully handle known errors... such as may occur if the system refuses storage or file updates for any reason.  This is part of many needed icing-on-the cake things.  I'll try to list a bunch of those in future issues just so they don't get forgotten, but as you develop I'm sure you'll run into many more possibilities I can't specifically spell out with forethought so please address them or note them in issues as you go.  If JavaScript has a good exception handling system, file operations (and others) are perfect for encapsulating with exceptions.  In the many years of the BASIC Stamp Editor's use, there are so many times where protections like that have helped our customers in the face of things we just don't have a whole lot of control over; they've expressed their appreciation for such things.  Thanks!

phated, 2015-04-08:
@PropGit correct, it is done in the manifest.

commit a260de80 (2015-04-08): add permission for unlimited storage - closes #56
commit ec96c59d (2015-04-08): Merge pull request #59 from parallaxinc/unlimited-storage

add permission for unlimited storage - closes #56
closed 2015-04-08 by phated