parallaxinc/Parallax-IDE #57: Can Chrome Web Store app be run on Chromium?

PropGit, 2015-04-06:
I tried running this on Chromium on a Raspberry Pi, but CWS seems to disallow it (because it's not Google's Chrome distro) and pointing Chromium at the unpacked extension fails to display the Details, Launch, and Reload links.

Is there a way to make it run in this case?

urbantumbleweed, 2015-04-09:
Hmm, it looks like I have it working on my end.  I wonder what version of Chromium you are using? That could be the issue as there where problems with google V8 and ARM processors that were only recently fixed. So, definitely be sure you have the latest chromium for the Raspberry Pi.

I didn't have any problems running this on mac.
Here is some information about my system at the time I was able to make it work.
Version 44.0.2362.0 (64-bit)
[image]

When I first opened chromium I was immediately presented with a link to the Chrome App Store at the top of the window near the toolbar.

[image]

I was also able to go to `chrome://extensions`, select 'developer mode' -> 
`load unpacked extensions` and `launch`.  I am also able to see the 'details' and 'reload' links.

[image]

Upon opening a new window I was able to see the application icon at `chrome://apps`.

PropGit, 2015-04-10:
Good point.  I'll check those things and update this thread when I can.

PropGit, 2015-08-20:
@urbantumbleweed - looks like Chromium in RPi is stuck at old versions. 

It shows Version 22.0.1229.94 Built on Debian 7.0, running on Debian 7.8 (161065)

Thinking there must be an updated build somewhere or that I could build it on the RPI, I started looking around and came up with the following Q&A's that lead to dismal thoughts:

https://www.raspberrypi.org/forums/viewtopic.php?t=51897&p=401514

http://raspberrypi.stackexchange.com/questions/31267/installing-latest-chromium-with-version-1

Any ideas?  Seems like a cross-compile would be necessary, but even so, may be very problematic because of limited run-time resources on the RPi.