parallaxinc/Parallax-IDE #60: Consider and Utilize Parallax Theme Colors

PropGit, 2015-04-08:
Since this is destined to be a "Parallax" branded development environment (company-centric rather than product-centric), we'd like to carefully consider color palette for it's skin.

We have a Parallax "theme" color palette that we use on our website and many product materials and we'd like to apply this where appropriate to this IDE.  Variations may be discussed, considered and applied depending on the overall effectiveness of the below color scheme.

Here are the hex and RGB equivalents.

__Main blues:__
Navy Blue (#002942) R:0 G:41 B:66
Medium Blue (#1474B7) R:20 G:116 B:183

__Gray options:__
Dark (#333333) R:51 G:51 B:51
Medium (#666666) R:102 G:102 B:102
Lighter (#999999) R:153 G:153 B:153
Lightest (#EEEEEE) R:238 G:238 B:238 

__Additional colors used on the eCom site:__
Yellow (#F5D63E) R:245 G:214 B:62
Orange (#FF9900) R:255 G:153 B:0
Red (#E1261C) R:225 G:38 B:28
Green (#8BBD49) R:139 G:189 B:73

urbantumbleweed, 2015-06-19:
Parallax IDE will execute branding to meet corporate guidelines while optimizing user experience and accessibility. This palette can be used as input for the design process.

jimjeffers, 2015-11-25:
@PropGit Thanks this wasn't apparent to me. I'll utilize these colors in  the UI mockups!