parallaxinc/Parallax-IDE #66: Fast data causes performance issues and apparent lock-up

PropGit, 2015-04-11:
Sending data faster than a certain rate (and maybe when combined with more than a certain number of received lines) causes v0.2.0 to slow down to a crawl.

On my computer, this code started causing the behavior around loop 250 and get's really bad around loop 350.  It appears to be a cumulative effect on display refresh rate and CPU usage.  The app appears to lock up, but it's actually very busy- you can power off or hold the reset button down on the BASIC Stamp board and eventually the app will return to life.

```
'{$STAMP BS2}
'{$PBASIC 2.5}

Idx VAR WORD

Delay CON 10

Idx = 0

DO
  Idx = Idx + 1
  DEBUG "Loop ", DEC Idx, CR
  HIGH 0
  DEBUG "On",CR
  PAUSE Delay
  LOW 0
  DEBUG "Off",CR
  PAUSE Delay
LOOP
```

PropGit, 2015-04-27:
Here's a video demonstrating this problem.  https://drive.google.com/a/parallax.com/file/d/0B0tev9Y3WaL4SEFmSlRYNEg2Njg/view?usp=sharing

PropGit, 2015-05-15:
Here's another video that demonstrates [Debug Throughput at 38.4 kBaud](https://drive.google.com/file/d/0B0tev9Y3WaL4NzZWM0lfeWRkSjA/view?usp=sharing).

closed 2015-05-22 by phated

2fast2fourier, 2015-05-22:
We just landed another change that reduces the performance penalty when running a very-fast debug output loop (<10ms).

The change rate-limits the expensive HTML update portion of console output. Currently, if data events arrive at a very high speed, the console will attempt to display each update as it comes in. With this change, we will throttle events if they occur too quickly, but ensure that a refresh will still occur within the throttle window (64ms).

This was specifically designed to avoid losing events during the throttle, if two debug statements appear within a few milliseconds of each other it will still show the second after the throttle window even if no other debug messages occur. The throttle window was set to 64ms in order to work best on low-powered chromebook devices, but can be set a bit lower if a faster update rate is needed. CPU usage also dropped considerably on laptops/desktops.