parallaxinc/Parallax-IDE #67: PBASIC Tokenizer Failures

PropGit, 2015-04-11:
I think there is some kind of problem in the delivery of the source code to the tokenizer.

NOTE: Plug in a BASIC Stamp, but DO NOT POWER IT for the following tests

1. Copy and paste the code below into the editor.
2. Attempt to Download
    * Error: 156- Unknown target module
    * The code is actually syntactically correct, this is a bogus error... continue with the steps below
3. Make the code a tiny bit shorter by removing the last `0` on the line `Delay CON 100`
4. Attempt to Download
    * Successful compile (failure to download is expected; we left the Stamp turned off... this is dangerous code)
5. Place your cursor after `LOOP` and press the enter key so that there are now 22 lines of code
6. Attempt to Download
    * Error: 156- Unknown target module
    * This is a bogus error again... continue with the steps below
7. With your cursor on line 22, press the enter key again so that there are now 23 lines of code
8. Attempt to Download
    * App freezes and CPU usage gradually increases more and more.  Need to kill it.
    * The code is syntactically correct, this further indicates source delivery problem

These results indicate a problem with source code delivery and/or source buffer allocation, causing the tokenizer to fail to properly parse and tokenize the code.

```
'{$STAMP BS2}
'{$PBASIC 2.5}

'This program causes Chrome IDE v0.2.0 to come to a halt

Idx VAR WORD

Delay CON 100

Idx = 0

DO
  Idx = Idx + 1
  DEBUG "Loop ", DEC Idx, CR
  HIGH 0
  DEBUG "On",CR
  PAUSE Delay
  LOW 0
  DEBUG "Off",CR
  PAUSE Delay
LOOP
```

PropGit, 2015-04-11:
Found a programming clue:  The source code above is exactly 256 bytes in length (when assuming single-byte characters, new lines and end-of-source marker).

PropGit, 2015-05-12:
I just saw the pull request on PBASIC-Tokenizer.  I'm saying "Yes, yes, yes!  Thank you!"  I figured it was something like that!  Keeping my fingers crossed now.

phated, 2015-05-14:
@PropGit we are getting these merged today.  Do you think we should do a v0.2.1 to get this bugfix in a published version?

jeffam70, 2015-05-15:
Oh wow, yes please!  That would be fantastic!

phated, 2015-05-15:
This has been released as 0.2.1 - https://github.com/parallaxinc/ChromeIDE/releases

Closing.

closed 2015-05-15 by phated