parallaxinc/Parallax-IDE #70: Keyboard Shortcut Interactions

PropGit, 2015-04-16:
Do you know if Chrome OS keyboard shortcuts override/collide with a Chrome App's shortcuts, or vice versa?

urbantumbleweed, 2015-05-19:
With most of the keyboard shortcuts completed I have at least a tentative answer that seems correct in most situations regarding your question on overriding/colliding.  Most of the shortcuts you have sent to us do not overlap with those defined for ChromeOS so we been significantly obstructed yet.  We have found that in some cases you need to enable or disable default behavior. Other keyboard combinations, namely those that include ‘Tab’, have limited functionality by default for accessibility reasons.

In ChromeIDE we deal with some overrides and force the app to take precedent by explicitly preventing browser default behavior.  We have yet to test whether this approach will work on ChromeOS.  So far we have found that our browser-based testing environment is not the same as the App, and we tend to encounter less keyboard mapping conflicts in the App; this is promising.  We have yet to complete our testing on the Chromebook though so I’ll need to get back to you.

Some resources that might help satisfy your curiosity.
[ChromeOS keyboard shortcuts](https://support.google.com/chromebook/answer/183101?hl=en)
[ChromeApp command API](https://developer.chrome.com/extensions/commands)

PropGit, 2015-05-19:
Thanks @urbantumbleweed.  Hopefully we don't run into a big problem that is difficult to resolve.  The Tab issue concerns me.

urbantumbleweed, 2015-05-19:
Please see the response on #88

closed 2015-05-19 by urbantumbleweed