parallaxinc/Parallax-IDE #73: Support for BASIC Stamp Identification and display

PropGit, 2015-05-08:
Add function to perform _only_ a BASIC Stamp Identification, and display the progress and results.

The Identification is achieved as the first part of the download protocol (ie: most of the work has already been achieved at the protocol level to enable this feature) but the process simply needs to stop short of the actual download.  The download protocol documentation describes what to do for the Identification only feature.

PropGit, 2015-05-15:
Here's a video demonstrating the [Identification display](https://drive.google.com/file/d/0B0tev9Y3WaL4UDRpWkxaU2xoRE0/view?usp=sharing), which also serves double-duty as the Download progress display.

urbantumbleweed, 2015-05-19:
Can you provide feedback about what is a reasonable timeout period for this process?

urbantumbleweed, 2015-05-19:
### Background
In the application we anticipate using scanning the for the presence of several board types. Additionally, on a computer there are potentially many ports we need to scan.

Currently we have set a 1 second timeout to search for a board on a port.  This translates to 1 second per board/port possibility.  This delay period will grow as there are more ports to search and additional boards need to be scanned.  To put in perspective, 30 seconds seems like a long time to scan for 5 different boards on 6 ports (30 possibilities).

### Question:
**Is there a more reasonable delay period we should use with the timeout?**

PropGit, 2015-05-19:
Please explain how the timeout is applied?  Is it a drop-dead timeout that overrides and terminates the search process for that particular port?  Is it meant to prevent app lock-up, or is it really meant to apply to waiting for a response from the board?

If it's meant to apply to waiting for a response from the board (of which there are many possibilities) it does indeed need to be as short as possible without being so short that it causes _false negatives_ on certain machines that just run slow.

There's really two factors to employ here, the no-echo termination and the timeout termination.

__No-Echo Termination__
The BASIC Stamp protocol itself relies on an echo (because the BASIC Stamp's hardware ties Tx and Rx together through a 4.7 kOhm resistor) so for each port that is opened, the process of scanning for BASIC Stamps should end _immediately_ if there is no echo detected.  In other words, if there is no echo detected when issuing the first parts of the protocol for the first BASIC Stamp it tries to discover on that port, it can then safely terminate and stop scanning for _any more_ BASIC Stamps on that port because there's no way it will ever find any in that case.  This is one of the things we do in BSE and it works very well to speed things up on ports that don't actually have a BASIC Stamp connected.

__Timeout Termination__
As implemented in BSE, the timeout is applied to waiting for a response from a board.  That response should occur very quickly (we're talking microseconds to perhaps a few milliseconds) but the computer's OS will slow things down in terms of notification and access to data in the OS's "software" Rx buffer and such.

We have this timeout (which is the amount of time to wait for a response byte from the BASIC Stamp _after_ receiving the echo) set to 40 milliseconds.

With this, we are able to scan for every possible BASIC Stamp (all 5 of them) on a particular port in about than 1 second.  See screenshot below showing break condition and transmitted hello code (Host Tx), plus echoed response (Host Rx) on a port with a BASIC Stamp connected but not powered.

[image]

Note: When we get to communicating over Wi-Fi, we'll have to employ a _different_ timeout that's specific to that medium.

closed 2015-05-21 by phated

PropGit, 2015-05-30:
Can not get Chrome IDE to recognize my BASIC Stamp any more.  It seems to hold the port open also; the BASIC Stamp Editor is unable to open the port after Chrome IDE has touched it.  Need to remove and reconnect the board to allow the BASIC Stamp Editor to open the port.