parallaxinc/Parallax-IDE #75: Debug Terminal: Support for all defined characters

PropGit, 2015-05-08:
Add support for all defined characters (visible and control characters) as defined in the __Transmit Pane__ and __Receive Pane__ sections of the [Debug Terminal](https://docs.google.com/document/d/1u0L0WTVU-jfvN4H56efHdBhNYTbObEX72Bn7Fc2RxwE/edit?usp=sharing) document.

PropGit, 2015-06-04:
NOTE: The Bell control character code (7) will not be implemented at this time due to issues on different platforms.  ASCII code 7 will simple be parsed and ignored (no effect) until possibly a later time if this can be resolved.

PropGit, 2015-06-10:
NOTE: The [Debug Terminal](https://docs.google.com/document/d/1u0L0WTVU-jfvN4H56efHdBhNYTbObEX72Bn7Fc2RxwE/edit?usp=sharing) document has been updated as of yesterday (see bottom of file) to give a better overall description of the auto-scrolling or not-auto-scrolling moments.

phated, 2015-06-12:
Currently, all but cursor-position, cursor-position-x, cursor-position-y and collapsing CR,LF are implemented. Those will be addressed after the underlying filesystem architecture is completed, as it is blocking some UI work.

closed 2015-06-16 by phated