parallaxinc/Parallax-IDE #76: Debug Terminal: Add Transmit Pane

PropGit, 2015-05-08:
Add Transmit pane to support host-to-microcontroller transmissions as defined in the __Transmit Pane__ section of the [Debug Terminal](https://docs.google.com/document/d/1u0L0WTVU-jfvN4H56efHdBhNYTbObEX72Bn7Fc2RxwE/edit?usp=sharing).

For BASIC Stamp's PBASIC language, this feature supports instructions DEBUGIN and SERIN.

drquinn, 2015-07-09:
@PropGit  - as I implement this transmit pane, I have a few questions about how it behaves that I did not see answered in the issue or Debug Terminal document. Sorry if you already answered these elsewhere...

1. What happens if a user types text into the transmit pane and no device is connected? Can they type text? Do they see an error message? Then when a STAMP is connected and a program is run, is the text cleared? Or should the entered text be sent character by character to the STAMP?

2. Is the transmit pane text area editable? Or is each character just added on to the end of a read-only list? It seems users should at least be able to use backspace, as you mention with the control character ASCII 8? But can they type in the middle of already entered text? I have seen this done a few different ways in different serial monitors, so I wanted to clarify what exactly you are looking for.

PropGit, 2015-07-09:
@drquinn - Good questions.

  1. We do have a defined behavior here, but I neglected to document it.  
    * If the Debug Terminal _is not_ currently connected to any communication port - the system does not allow the user to use the transmit terminal at all; they can't even click in it (no blinking cursor, no focus capability) it is disabled and visibly indicates that.  In the case of the BSE's Debug Terminal, it looks like this during this situation:
        
[image]

Note that there's text in the Transmit Pane left over from a previous connection, but now the port is "NONE" meaning it's not connected to any communications port.  We don't currently have a visible port field for Parallax IDE's Debug Terminal, but the concept still applies that sometimes it's simply not connected to a port.

  * If the Debug Terminal _is_ connected to any communication port - the Transmit Pane is enabled and the user is free to click on it and type text and the system transmits that text (character by characters as it receives it real-time) to the connected port at the current baud rate.  It doesn't matter whether or not a BASIC Stamp is actually connected to the port or not- this behavior applies regardless of that.  This looks like the image below in our current BSE DT design:

[image]

  * Whenever a new download occurs on that port (a new PBASIC program is sent to the connected Stamp) the system should clear both the Transmit and Receive panes of the Debug Terminal.

  2. The user should be able to scroll back and forth through the Transmit Pane's history, move the cursor around and select, copy, and paste text (and use the backspace key to delete the most recently typed characters); however, any text entry (via typing on the keyboard or via paste operation) should just append text to the right of the last visible text in the Transmit Pane.  When an entry or paste operation occurs, if the cursor happens to be anywhere except just after the last text in the pane, the cursor should first jump to the end of last text in the pane and then perform the entry or paste operation.  In truth, our current implementation doesn't force this last part, but it's what we'd like to implement going forward.

drquinn, 2015-07-10:
Thanks @PropGit, those answers were perfect. I believe I was able to get everything you asked for implemented.

During testing, I noticed that each key I enter into the Transmit Pane is displayed in the Receive Pane. Is this expected?

PropGit, 2015-07-10:
@drquinn - You're welcome.

> I noticed that each key I enter into the Transmit Pane is displayed in the Receive Pane. Is this expected?

Darn, I forgot to write about that too.  Yes, that's expected because of the way the BASIC Stamp is wired; it causes an automatic electronic echo of what was sent to it.

We need to have an Echo Off feature (which is enabled by default; ie: echo is off by default).  Since echo is _always_ caused by the BASIC Stamp, the software has to purposely look for and remove echos if the Echo Off feature is enabled.  I'll make an issue for it.

closed 2015-07-15 by phated

PropGit, 2015-07-24:
Verified added.  Experienced some problems.  Will note in new issues.

PR #183 (2015-07-15): Transmit pane
#### What's this PR do?
Implements a transmit pane to support host-to-microcontroller communication, as defined in issue #76 and the attached document.

#### Where should the reviewer start?

Pull down this branch, remove node_modules, npm install - to get the latest updates to bs2-serial.
npm run build

#### How should this be manually tested?
The Transmit Pane is the new bar between the editor and the Receive Pane. Right away, you should not be able to click into the Transmit Pane because no devices are connected. Load in some test code that uses SERIN. Download the code to a B [... cut]