parallaxinc/Parallax-IDE #77: Debug Terminal: User-settable baud and port

PropGit, 2015-05-08:
Add port and baud rate settings for the Debug Terminal.

The port to which the terminal is attached, and its baud rate, should be indicated somewhere near the terminal itself.  The values are adjustable by the user at any time.

More information may be read in the __Settings__ section of the [Debug Terminal](https://docs.google.com/document/d/1u0L0WTVU-jfvN4H56efHdBhNYTbObEX72Bn7Fc2RxwE/edit?usp=sharing) document.

urbantumbleweed, 2015-06-18:
This issue has only minor implications on UX decisions.  It could be related to issue #160 depending on the implementation.  Because the this largely an implementation issue, the UX/UI label is removed.

urbantumbleweed, 2015-10-06:
This issue may need to be split into two issues.

PropGit, 2015-10-06:
Agreed.