parallaxinc/Parallax-IDE #78: Debug Terminal: Tx/Rx Indicators

PropGit, 2015-05-08:
Add Tx and Rx activity indicators for the Debug Terminal.

The Tx (transmit) and Rx (receive) indicators should be somewhere near the terminal itself.  They each individually flash once for every block of data transmitted or received.

More information may be read in the __Settings__ section of the [Debug Terminal](https://docs.google.com/document/d/1u0L0WTVU-jfvN4H56efHdBhNYTbObEX72Bn7Fc2RxwE/edit#) document.

urbantumbleweed, 2015-06-11:
@PropGit I wanted to reach out to you regarding how the RX/TX indicator elements are to be implemented.  There seems to be some questions about how this element is to appear.  Could you provide us with a note about it's role in the user experience and your thoughts about it might look or where it should go?  Screenshots, mocks or wireframes would be excellent.

If there are any further questions I'll let you know.

PropGit, 2015-06-11:
@urbantumbleweed I'm somewhat open to suggestions here, but we definitely need to open up some area on screen where this and a few other terminal-related controls can reside.  Somewhere that is visible only when the Debug Terminal is visible.

Here's a crude mock-up of one possibility.  The circles next to TX and RX are like LED lights.  They each should light up when there's activity.

[image]

The point of these is to provide a quick visual indication that some data was received or transmitted as there are many cases when data can move but there's no visual change to the textual display in the Debug Terminal.  For any data movement event, the indicators should light up for a minimum of about 50 ms so that the user can see that something happened (even if it was just one character received at 3 Megabaud that cause it to light).

On our development boards, we use blue LEDs for Rx and red LEDs or Tx.  I think using this color scheme for this case would be good for consistency.

phated, 2015-06-15:
Completed by #147.  Will need to add some functionality to this when we implement DEBUGIN functionality, but that should be covered in that issue.

closed 2015-06-15 by phated