parallaxinc/Parallax-IDE #79: Debug Terminal: Automatic (event-driven) port arrival/departure handling

PropGit, 2015-05-08:
Gracefully handle USB-based Virtual COM Port arrivals and departures.  

The legacy serial port support that is also used for USB-based Virtual COM Ports may act in undesirable ways if the application has the port open when unexpectedly is it physically disconnected (causing the port to disappear from the system).  On Windows, this often resulted in preemptive process lock-out.

  1) Test what happens on various supported platforms when an open port is physically removed
  2) If the result is unfavorable, see if there's a way to resolve it like what we did in the BSE.

More notes can be found in
  * the __Settings__ section of the [Debug Terminal](https://docs.google.com/document/d/1u0L0WTVU-jfvN4H56efHdBhNYTbObEX72Bn7Fc2RxwE/edit?usp=sharing) document.
  * the __Communications > Serial/USB Ports > Automatic (event-driven) port refresh__ section of the [Chrome IDE Phase 2 Discussion Notes](https://docs.google.com/document/d/1lKBstygUt17s0X4s2q4nKF9Zc8pQdPDV-Zq6QeNqZ5A/edit?usp=sharing) document.

PropGit, 2015-05-12:
UPDATE:  Testing on Windows.

On Win 7 64-bit, when Chrome IDE has downloaded and is successfully receiving data from the BASIC Stamp, disconnecting the USB port doesn't cause a visible problem immediately; however, the port remains open and presumably Chrome IDE's handle is invalid.

  * Disconnecting USB port while in use causes no visible issue
    * Chrome IDE still functions and can download to other ports
  * Reconnecting USB port doesn't result in Debug Terminal display working again
  * Trying to download again to that port fails with a message indicating Chrome IDE can't open the port
    * Perhaps, when a user tries to download, a simple call to close the handle first, then a call to open the port would fix this?
  * Closing Chrome IDE and reopening Chrome IDE allows it to download to the port once again

Though this is a better situation than what we had experienced in BSE on Windows upon active port departure, we still need Chrome IDE to handle it more gracefully than it currently does by automatically handling the situation as described in the Debug Terminal document (noted in prev post).

The idea to tap into USB events and then re-enumerate the existing COM port list is a good one that will fit well as part of this solution.

PropGit, 2015-10-06:
Here's a list of the VID/PID pairs we are interested in monitoring for spontaneously departure/re-arrival:
```
   VID   /  PID   - Device 
  0x0403 / 0x6001 - FT232AM/BM/R, FT245R
  0x0403 / 0x6015 - FT231X
```

PropGit, 2015-10-06:
Reminder, the algorithm for handling departure and arrival events is listed in the third major bullet of the __Settings__ section of the [Debug Terminal](https://docs.google.com/document/d/1u0L0WTVU-jfvN4H56efHdBhNYTbObEX72Bn7Fc2RxwE/edit?usp=sharing) document.  Any mention of "visible port field" can be treated as "the current known port in use" since we don't currently have a visible port field.

phated, 2015-10-16:
The comment at https://github.com/parallaxinc/Parallax-IDE/pull/299#issuecomment-148805652 outlines a caveat on Windows for this feature.  However, it will work on OSX and ChromeOS due to no reliance on WinUSB on those platforms.

closed 2015-10-16 by phated

PropGit, 2015-10-22:
Verified:
  * __Works__ on Mac OS X
    * must be host; feature does not work on VM
  * __Works__ on Chromebook
  * __Works__ on Ubuntu
    * must be host; feature does not work on VM
  * __Does not work__ on Windows for reasons we can not fix

PR #299 (2015-10-16): Automatic Reconnect on USB discovery
#### What's this PR do?
This enables USB device detection, and automatically reconnects a recently-disconnected device if one is discovered. 
#### What are the important parts of the code?
The reconnect process is in the `deviceAdded` handler, and will only trigger if a device is selected and is available in the updated device list.
#### How should this be tested by the reviewer?
Unplug the device after bootload, then reconnect. Verify terminal output resumes as normal. Note: Reconnecting the device will reboot the board, causing any output to start from th [... cut]