parallaxinc/Parallax-IDE #82: Progress Display (during Identification or Run (download))

PropGit, 2015-05-10:
It's vital that the Chrome IDE display active progress status during Identification and Run (download) processes; without them, there's often no visual or audio feedback for the user.  This is also critical for diagnosis purposes as hardware connection issues can be elusive.

Examples of this active progress status are in the [Handling Multiple Stamps](https://drive.google.com/a/parallax.com/file/d/0B0tev9Y3WaL4Q3huMVhiMkJRdFU/view?usp=sharing) video.  This employs a dynamically changing list of existing ports, updating in real-time to show what microcontrollers (if any) were found on them (as well as other possible port status), a dynamic status message at the top of the screen and a progress bar at the bottom of the screen during an actual download.

PropGit, 2015-05-11:
Here's an [Identify and Run video](https://drive.google.com/file/d/0B0tev9Y3WaL4TjBxN2VsWFNPVGs/view?usp=sharing) that better demonstrates the kind of visual feedback and information were looking for during an Identification or Download process.

PropGit, 2015-05-15:
Here's a video demonstrating the [Identification display and later the Download Progress display](https://drive.google.com/file/d/0B0tev9Y3WaL4UDRpWkxaU2xoRE0/view?usp=sharing).

urbantumbleweed, 2015-05-19:
We need your guidance on the implementation of the progress display.  We have a couple option for you to consider.

### Options
1) Present port list that updated as process runs OR
2) Present an indeterminate progress bar  until the process is complete and show a complete port list.

### Considerations
Option 1 is more difficult to implement and will not be completed by the demo because it will require architecting event emission to update a view.
Option 2 is simpler to implement and could be completed by the demo on Friday.

### Question
**Are you ok with us doing the second option?**

PropGit, 2015-05-19:
This is an interesting situation.  The answer's yes and no.  

We much prefer Option 1 ideally, because it gives informed visual feedback for cases where there's an unexpected delay (sometimes caused by the OS negotiating with the USB device or installing the driver, or other strange behavior) and the rare but occasional case where the OS actually preempts and never responds because of some system issue (a hardware problem).  Without a display showing what it figured out before some unexpected hanging event, the user is left with no valuable feedback.

Is Option 2 something that could be implemented easily now and then improved upon to transform it into Option 1 in a later sprint?

If yes, let's go with Option 2 for now.  If no, let's do something super easy, and maybe ugly, for now, with the intention to continue work on this issue in Sprint# 5.

phated, 2015-05-21:
Completed by #100

closed 2015-05-21 by phated

PropGit, 2015-06-03:
Which option was implemented?  I can't tell.