parallaxinc/Parallax-IDE #84: Cloud (Google Drive?) support

PropGit, 2015-05-10:
Google integration is highly desired.  This would naturally allow the user to use their Google account to have real-time access to project files from multiple devices.

Some kind of Google-supplied, or seemingly-Google-supplied, cloud support is what is meant here; not necessarily Google Drive.  It's important that the cloud support be easy and natural so the user doesn't even have to think about it.  A cloud form of local browser storage is what's desired, attached to their Google account so that it's accessible (syncable) from multiple devices (running Chrome IDE) automatically.

This was discussed briefly in the __File Management > Importing/Exporting between drive/download/USB/local storage__ section of the [Chrome IDE Phase 2 Discussion Notes](https://docs.google.com/document/d/1lKBstygUt17s0X4s2q4nKF9Zc8pQdPDV-Zq6QeNqZ5A/edit?usp=sharing) document.

NOTE: Google Drive is also desired for backing up, exporting, importing, and sharing projects/source.  This, however, is a separate feature from this issue.  See [Issue 83](https://github.com/parallaxinc/ChromeIDE/issues/83) for more info.

urbantumbleweed, 2015-06-15:
@PropGit We have discovered that Chrome SyncFileSystem fails when the user is not logged into a google account.  On a positive note, we have confirmed there is offline support but the user needs to be logged in. For this sprint, Google syncFileSystem storage functionality will be added with the restriction that it will only work for users who are logged in.

This issue appears to be related to an existing issue.  You can find it [here](https://code.google.com/p/chromium/issues/detail?id=374101)

Is this a valid project constraint given the target platform of ChromeBooks that require login?  If not, we can discuss this further in another issue that is scheduled for a future enhancement.

PropGit, 2015-06-15:
@urbantumbleweed To answer your question, this is fine for now, but we'll need to address it again later.

I have some questions.
  * By "log in" do you mean to Chrome?  Or do we / can we provide a separate log in prompt ourselves at some point?
  * Does it fail in a graceful way that we can trap and silently handle or choose to prompt the user or not?
  * We are continuing to maintain things via indexdb just like we have been regardless of syncFileSystem, right?  So this doesn't really cause a big issue from a user perspective yet.
  * That's great that it has offline support.  Is it possible for the user to "log in" if the there's no Internet connection?  I think it is; my Chromebook seems to do so before it has Wi-Fi established.
    * So, that log on causes syncFileSystem to work automatically in it's offline mode?

PropGit, 2015-06-16:
> #6 kinuko@chromium.org
> I suspect this probably needs to be closed as WontFix, unfortunately.  (Assigning to tzik@ in case he has a plan)
> Status: WontFix 
> Owner: tzik@chromium.org 

Ouch... but after that amount of time, it's what I expected to.

We've got to be able to handle this outside of a deeper fix; I can't imagine at this point that it's impossible to deal with gracefully... just will have to be in another sprint.

phated, 2015-06-16:
That is quite disappointing, as the issue linked was supposed to be fixing the problem (according to the description).  It isn't impossible, but the amount of effort needed to solve this is significant.  I think trying to get Google to fix this is a good idea (probably by opening another issue that is exclusive to non-logged in support instead of multiple things).  SyncFileSystem should totally support this feature.

PropGit, 2015-06-16:
I agree, this is a real issue, but just how could it actually be solved by SyncFileSystem?

I mean, on a multi-user system, irregardless of how/where it stores the files during logged-out state, how would it associate the files to any person/account so that it knows how to resolve it once someone logs in?

urbantumbleweed, 2015-06-16:
## Answers in response to yesterday

* *By "log in" do you mean to Chrome? Or do we / can we provide a separate log in prompt ourselves at some point?*
 **That is correct, we mean Chrome.**

* *Does it fail in a graceful way that we can trap and silently handle or choose to prompt the user or not?*
  **Currently we are not handling the syncFIleSystem fail.  We need to consider options and spec our requirements in additional issues**

* *We are continuing to maintain things via indexdb just like we have been regardless of syncFileSystem, right? So this doesn't really cause a big issue from a user perspective yet?* 
 **Not exactly, with your decision to use syncFileSystem api we have replaced indexdb in all cases except for temporary files.**

* *That's great that it has offline support. Is it possible for the user to "log in" if the there's no Internet connection? I think it is; my Chromebook seems to do so before it has Wi-Fi established.*
 **Initial log in requires internet. The user must log in with an internet connection but once logged into Chrome the session persists even if internet connection is lost.  The session persists until the user logs out or the machine sleeps.  On a positive note, if the user has logged into Chrome on the machine before, subsequent log in is possible when offline**

* *So, that log on causes syncFileSystem to work automatically in it's offline mode?*
 **That is correct.**

* *When the user launches the app, how does it choose which log-in to use? If frequently have my Chrome browser logged into my personal address but I have a number of tabs open that are logged in to Google services with another account.*
 **There can only be one user logged into a Chrome instance at a time.  Therefore, the app uses the current user profile at the time it is launched. The logged in user is separate from the email addresses used to log into google services.**

You can see the user is "Guest" as indicated in the top right corner.
[image]

This is the view when switching users
[image]

We still need to follow-up with answers to your questions from your last comment.

phated, 2015-06-17:
@PropGit every chrome instance has some sort of "user" even if it is an anonymous user on first launch.  I don't think syncFileSystem would transfer any files saved on an anonymous user to a logged-in user's account, but it should at least gracefully degrade to the offline-only FileSystem API

PR #150 (2015-06-17): replace browserify-fs with chrome-sync-fs
#### What's this PR do?
Replaces `browserify-fs` with `chrome-sync-fs` during the bundle process.  This allows us to continue to use the workspace implementation as it currently stands, including temp file support using IndexedDB.  The workspace abstraction still needs work before it will be ready to ship.

#### Where should the reviewer start?
`npm install` and `npm run build`

#### How should this be manually tested?
Verify that the application still works (you might need to clear the user config).  Create some files, add some text, jump between file [... cut]
closed 2015-06-17 by 2fast2fourier