parallaxinc/Parallax-IDE #85: PBASIC Syntax Highlighting

PropGit, 2015-05-11:
Add PBASIC Syntax Highlighting as described in [Issue 71](https://github.com/parallaxinc/ChromeIDE/issues/71).

NOTE: True PBASIC Syntax highlighting also changes the case of the symbols in the source code, and also causes those changes to be saved in the file (as if entered that way by the user).  Since that is probably beyond what is readily available in the Chrome IDE edit control, we will gladly defer the case-modification feature for now and just request highlight and style adjustment.

monteslu, 2015-05-18:
@PropGit got the first pass at recognizing all the reserved words, handling indentation correctly in if-elseif-endif, do-while blocks, etc.  As well as correctly recognizing comment blocks.

[image]

I could use some advice on matching the colors described in #71 to hex colors so that we can update the styles in codemirror.  Is it possible for you to match the colors in the attached screenshot to what they need to be?

PropGit, 2015-05-19:
I'm not positive I know what advice you need, but I just posted more clear syntax metrics in post# 5 of Issue #71. 

In the colored example you provided,
  * the ```Rnd```, ```LED1```, and ```GetRandom:``` are a user-defined symbols, they don't match any reserved words, so the SynDefault gets applied (black font on white background, plain font style, no case change).
  * ```VAR``` and ```PIN``` are SynDeclaration.
  * ```Word``` is a SynVariableType.
  * ```0```, ```100```, ```999```, and ```1``` are SynConstantDecimal.
  * ```DO```, ```GOSUB```, ```HIGH```, ```PAUSE```, ```LOW```, ```LOOP```, ```RANDOM```, and ```RETURN``` are SynCommand.
  * ```=``` is SynConditionalOp.
  * ```//```, and ```+``` are SynBinaryOp.

closed 2015-05-19 by monteslu

PropGit, 2015-05-29:
Verified all reserved words are syntax highlighted correctly, except DEBUGIN... this is because I mistakenly labeled it as an etConstant.

PR #96 (2015-05-19): Syntax
#### What's this PR do?
Apply syntax colors as described in issue #71 
Targets PBASIC full syntax 
Closes #85 and #71 

#### Where should the reviewer start?
Load this branch and run ChromeIDE.

#### How should this be manually tested?
 Create a test file and enter in some test code. Try to use each of the token categories listed in issue 71 to ensure they are being properly colored.

#### Any background context you want to provide?
The highlighter is based on the VB code file. The css theme is modified from neo. 
Only piece not working as it should is the etDirective type.

#### What a [... cut]