parallaxinc/Parallax-IDE #88: Add Keyboard Shortcuts

PropGit, 2015-05-11:
For devices with hardware keyboards, keyboard shortcuts are critical to ease-of-use for the intermediate and expert developer.

This was discussed in the __Editor > Keyboard Shortcuts__ section of the [Chrome IDE Phase 2 Discussion Notes](https://docs.google.com/document/d/1lKBstygUt17s0X4s2q4nKF9Zc8pQdPDV-Zq6QeNqZ5A/edit?usp=sharing) document.

There are certain often used shortcuts, and a few others that are less used but very handy.  The list will change over time, but here's what we request right now.    NOTE: Some of these are not possible on Chromebooks, but are commonly used on other platforms.

__Editing__
  * Ctrl-A - Select all text in current source code
  * Ctrl-C - Copy selected text to the clipboard
  * Ctrl-F - Find or replace text
  * Ctrl-V - Paste text from clipboard to selected area
  * Ctrl-X - Cut selected text to the clipboard
  * Ctrl-Z - Undo last action (unlimited)
  * Ctrl-Shift-Z - Redo last action (unlimited)
  * Tab - Indent block (Inserts space characters)
  * Shift-Tab - Outdent block (Deletes space characters)
  * F3 - Find text again
  * Shift+F3 - Find text again (opposite direction)
  * F4 - Replace current found selection
  * Ctrl-F4 - Perform replace and find next
  * Ctrl-Home - Jump to top of file
  * Ctrl-End - Jump to end of file
  * Ctrl-CursorUp - Move source view up one line without moving cursor
  * Ctrl-CursorDown - Move source view down one line without moving cursor
  * Ctrl-+/- (or =/_) - increase/decrease font size

__File Management__
  * Ctrl-O - Open source; may just focus file explorer
    * Cursor + enter keys navigate and select files
    * Backspace exits file view (project) (to project view)
  * Ctrl-S - Save current source code file to its current location on disk
  * Ctrl-P - Print current source
  * Ctrl-Tab - Switch to next open source
  * Ctrl-Shift-Tab - Switch to previous open source

__Code__
  * F6 or Ctrl-I - Identify BASIC Stamp firmware.
  * F7 or Ctrl-T - Perform a syntax check on the code.
  * F8 or Ctrl-M - Tokenize and open Memory Map display.
  * F9 or Ctrl-R - Tokenize, download to BASIC Stamp, and open Debug display if necessary.
  * ESC - Close current dialog.

PropGit, 2015-05-11:
This relates to a question, in [Issue 70](https://github.com/parallaxinc/ChromeIDE/issues/70).

phated, 2015-05-19:
I believe that most/all of the functionality that we currently have in the IDE is mapped to the correct key commands now.  Leaving for @urbantumbleweed to list and verify the implemented key commands below and then close this issue.

urbantumbleweed, 2015-05-19:
##Update:
 There are 6 keystroke combinations from the list above that are problematic on a ChromeBook. I have not yet verified that valid combinations can invoke a handler without other side-effects; that is the next step. 

The 6 combinations that aren't working are:
CTRTL-Home - The home key does not exist and needs an alternative, maybe CTRL-'
CTRL-End - The End key does not exist and needs an alternative, maybe CTRL-/
CTRL-+ - This is a ChromeOS combination binding to control window size
CTRL-_ - This is a ChromeOS combination binding to control window size
CURSOR+ENTER - The 'Enter' key is not a modifier so key events are not registered together so this combination is not possible.  We are assuming that CURSOR means UP and DOWN.
CTRL-T - This combination is used to create tabs and does conflict.
CTRL-W - This combination will close the window

TAB was registering during the preliminary test.

The next step will be verify that valid combinations behave as expected with handlers are attached.  No unintended side-effects.

PropGit, 2015-05-19:
[CURSOR+ENTER - The 'Enter' key is not a modifier so key events are not registered together so this combination is not possible. We are assuming that CURSOR means UP and DOWN.]

You must be talking about "Cursor + enter keys navigate and select files" ?

If so, that's my fault for being unclear.  I meant, the UP and DOWN (cursor) keys visibly highlight files in the Project/Files view, and then the Enter key can be pressed to select (ie: open) that file/folder.  Also, the LEFT and RIGHT (cursor) keys can be used to enter or exit a folder.

Some of this is functionality relates to a future issue/sprint as it deals with Project/File view and exploration, but some may be implemented now if it all works out easy.

The main point is, there should not be any elements or features on-screen (with some minor exceptions) that cannot be accessed by keyboard (ie: almost none of the features should _require_ exclusive mouse action; for devices with a mouse).

PropGit, 2015-05-19:
__[CTRL-T - This combination is used to create tabs and does conflict.]__

Can we override it?

__[CTRL-W - This combination will close the window]__

Where did I define Ctrl-W?

PropGit, 2015-05-19:
__[CTRL-+ - This is a ChromeOS combination binding to control window size
CTRL-_ - This is a ChromeOS combination binding to control window size]__

This appears to control the zoom level of the browser contents on my Chromebook.  Is that what you meant?

It doesn't seem to do anything when ChromeIDE is active... is it really causing a problem?  Can we override it?

If we can't override it (or it causes some problem) let's use Ctrl-Alt- -/+ (or _/=) depending how you look at it (no shift key involved though).

PropGit, 2015-05-19:
__[CTRTL-Home - The home key does not exist and needs an alternative, maybe CTRL-'
CTRL-End - The End key does not exist and needs an alternative, maybe CTRL-/]__

The solution to this is (from an earlier email):

I assume you mean that the HOME and END key shortcuts will still work on Win/Mac/Linux, but we need to find something else that gets substituted in when run on Chromebook.  Is that right?

For _jump-to-top-of-file_ and _jump-to-end-of-file_ functionality, let's substitute in __Ctrl+Search+Left__ and __Ctrl+Search+Right__ just like defined by Chromebook's keyboard shortcuts.

urbantumbleweed, 2015-05-19:
It turns out that we in my tests I needed to prevent default behavior.  There are currently no issues with for the following combinations.
CTRL+SEARCH+LEFT
CTRL+SEARCH+RIGHT
CTRL-+
CTRL-_
CTRL-T

My apology, the CTRL-W is not listed and I was testing it as a known binding in ChromeOS.  It too is now being overridden in our tests.

CURSOR followed by ENTER is not currently implemented. How will this affect the cursor movement inside the editor window?  Does navigating files (UP/DOWN + CURSOR) and enter/exit folder (LEFT/RIGHT) take precedent?

PropGit, 2015-05-19:
Oh good.  No problem.

For CURSOR followed by ENTER sequences, that only applies when the Project/File view has focus; otherwise, CURSOR and ENTER keys do the normal editing thing.

urbantumbleweed, 2015-05-19:
Here is the status on the individual keyboard shortcuts.

### Editing
**works** Ctrl-A - Select all text in current source code
**works** Ctrl-C - Copy selected text to the clipboard
**works as Find** Ctrl-F - Find or replace text
**works** Ctrl+Shift+F - replace text
**works** Ctrl-V - Paste text from clipboard to selected area
**works** Ctrl-X - Cut selected text to the clipboard
**works** Ctrl-Z - Undo last action (unlimited)
**works** Ctrl-Shift-Z - Redo last action (unlimited)
**works for single line** Tab - Indent block (Inserts space characters) -
**broken for multi-line** Tab - Indent block (deletes content)
**works for single and multi-line, brings tab to first character position** Shift-Tab - Outdent block (Deletes space characters)
**work with SEARCH+3** F3 - Find text again
**works with SEARCH+SHIFT+3** Shift+F3 - Find text again (opposite direction)
**not mapped - need clarification** F4 - Replace current found selection
**not mapped** Ctrl-F4 - Perform replace and find next
**replaced with Ctrl+Search+Left** Ctrl-Home - Jump to top of file
**replaced with Ctrl+Search+Right** Ctrl-End - Jump to end of file
**works** Ctrl-CursorUp - Move source view up one line without moving cursor
**works** Ctrl-CursorDown - Move source view down one line without moving cursor
**not mapped** Ctrl-+/- (or =/_) - increase/decrease font size
### File Management

**works** Ctrl-O - Open source; may just focus file explorer
**not mapped** Cursor + enter keys navigate and select files
**not mapped** Backspace exits file view (project) (to project view)
**works - quirk when saving on new blank file** Ctrl-S - Save current source code file to its current location on disk
**works - but needs to isolate code content** Ctrl-P - Print current source
**works - once a file is selected** Ctrl-Tab - Switch to next open source
**works - once a file is selected** Ctrl-Shift-Tab - Switch to previous open source
### Code

**function not completed yet- should work** F6 or Ctrl-I - Identify BASIC Stamp firmware.
**function not completed yet- should work** F7 or Ctrl-T - Perform a syntax check on the code.
**function not completed yet- should work** F8 or Ctrl-M - Tokenize and open Memory Map display.
**function not completed yet- should work** F9 or Ctrl-R - Tokenize, download to BASIC Stamp, and open Debug display if necessary.
**works** ESC - Close current dialog.

I am also including new mappings that we need to implement as backup options to F# keys.
> For Win/Mac/Linux (ie: legacy keyboard devices) we still want those function keys to work as defined.  > For Chromebook however, there's no single-key presses we could use, so we'll stick to double-key  combinations defined by Issue #88 (F6-F9) or those defined for Chromebook already (if they exist) and triple-key combinations I show below.

> "F3 - Find text again" becomes Ctrl+G (defined by Chromebook)
> "Shift+F3 - Find text again (opposite direction)" becomes Ctrl+Shift+G (defined by Chromebook)
> "F4 - Replace current found selection" becomes Ctrl+Alt+F (defined by me)
> "Ctrl-F4 - Perform replace and find next" becomes Ctrl+Alt+G (defined by me)

PropGit, 2015-05-19:
Does Tab / Shift+Tab insert/remove 2 space characters?  If not, we'd like it to insert spaces to line up with absolute multiples of 2 characters (column 3, 5, 7, 9, 11; assuming leftmost column is column 1).

urbantumbleweed, 2015-05-19:
Although multi-line functionality is partially working, as a development is considered a new issue (it wasn't spec'd) and will be addressed after the demo.  

Shift-Tab removes 2 spaces as expected on Mac but is collapsing to column 1 in Chrome.  'ChromeOS support' will become a label so we can address this and other similar cases new issues.

closed 2015-05-19 by urbantumbleweed

PropGit, 2015-05-29:
__Ctrl-Tab and Ctrl-Shift-Tab__
As implemented in v0.3.0, it's not exactly what I indended.  The intent was to provide navigation to the next or previous file recently opened.  In other words, imagine the app is a multi-tabbed browser, but without the tabs.  But even without tabs, the Ctrl-Tab and Ctrl-Shift-Tab combinations switch the edit view to the next or previous source that is opened.