parallaxinc/Parallax-IDE #91: File/Project View Re-envisioning

PropGit, 2015-05-11:
Let's consider and discuss our target UX design.

I'd like us to change how projects and files are accessed, viewed, and manipulated to make it more device agnostic.  

We should take the existing project view pane (left side of app in v0.2.0), collapse it (out of the way) and move it to appear from the top of the app (title/button bar).  Also, we should display the project and object title, which makes it obvious what source code is currently being viewed and also serves as a touch target for access the project/file view.  The project/file view also houses the project and file manipulation features.

A [mock-up video](https://drive.google.com/file/d/0B0tev9Y3WaL4djRtNGVkS3Z4ZWM/view?usp=sharing) demonstrates this idea as well as a few other UX items.

urbantumbleweed, 2015-06-11:
@PropGit Thanks for the video. You introduced a number of ideas that with your video.  I really like what you've done and where your head is at. Would you be able to break out the content of the video into written specifications and corresponding issues to discuss individually? I can work with the team to then group them into sprints and work them into discussions with our designer.

PropGit, 2015-06-11:
@urbantumbleweed You're welcome, and thanks.

Yes, I can do that.  I will try to provide some more thorough descriptions and updated mock ups of individual features.  How soon do you expect to need this for discussion among yourselves or with a designer?  I'm thinking I can have it all described and entered into issues by early next week at the earliest.