parallaxinc/Parallax-IDE #106: File dialog should set focus on filename field automatically

PropGit, 2015-05-29:
The New File dialog should put focus on filename field automatically once opened.  

Request: A very slight delay (after dialog appearance) followed by a focus-event animation (perhaps an expanding and translucent-fading circle on the field) to simulate the appearance of a user click/touch on the field, followed by the existing "filename" label slide-shrink animation.

If it's possible for a previous entry to appear (like what happens after an Esc key press on a previous dialog, then the cursor should appear at the end of the field's filename text.

urbantumbleweed, 2015-06-09:
@PropGit What is meant by "If it's possible for a previous entry to appear (like what happens after an Esc key press on a previous dialog), then the cursor should appear at the end of the field's filename text."  Do you want to have the previous text to appear even in the even if the user clicks cancel instead of pressing ESC?

urbantumbleweed, 2015-06-09:
@PropGit Also, the animation request hasn't been implemented yet.  We are thinking that it could be separated out into another issue that is grouped with UI/UX.  The final behavior may become standard visual language on the website.  What do you think about moving it to another issue that we label UI/UX?

PropGit, 2015-06-10:
@urbantumbleweed Sorry, I was writing that fast and should have re-thought about it before submitting.

I'd rather that a Cancel or Escape event cause any text changes the user made in the filename field to be undone and forgotten (not persistent).  

Note: This dialog has a related issue #107 that is also in Sprint 5, which affects what appears in the filename field upon displaying the dialog.

I agree to moving the animation request to another issue, but can the "existing 'filename' label slide-shrink animation" remain?

urbantumbleweed, 2015-06-11:
Thanks for the response.  I will go ahead and create the related issue.  The existing animation will remain.

PR #145 (2015-06-12): Focus
#### What's this PR do?
Sets focus in the new file modal to the 'filename' input when creating a new file. 

Esc key will clear anything in the 'filename' input.

#### Where should the reviewer start?
Load changes, open ChromeIDE.

#### How should this be manually tested?
Click the new file button. The focus should be on the 'filename' input for you to immediately start entering text. 
Click cancel. 
Click the new file button again. The text that was previously in the input should be gone.
Enter new text.
Press the esc key.
Click the new file button again. The text that was previously in [... cut]
closed 2015-06-12 by phated