parallaxinc/Parallax-IDE #107: New source function

PropGit, 2015-05-29:
__Problem__
Currently, entering source in the first session's blank editor presents no problem, until you go to save it, which presents an "EISDIR, illegal operation on a directory '/new-project' error.

__Solution__
The common workflow in BSE and PT does not require a user to create (and decide on a name for) a new source code file before entering source code in the editor (although it's an option they can choose to do).

  1. Change New File function to simply clear the editor (do not display the filename dialog) and assume a default filename of "untitled#" where "#" is a numeric value starting at 0 (for each new session) and increasing for each selection of New File.  NOTE: Assume the file will be saved in the current project folder, but do not force it to be.  It's best if it's created/treated in a special way that allows it to remain easily accessible to the user (during this session or the next) but for the system to know that it has never really been "saved" by the user.
  2. Selecting the Save File function now behaves like a Save As function; it displays the filename dialog (that used to be displayed by the New File function) and is pre-populated with the default filename "untitled#" with the entire filename selected so that any keyboard entry replaces it.
    * Note: Or the default filename can appear in the field but grayed-out and disappears upon the entry of the first character from the keyboard.

drquinn, 2015-06-10:
@PropGit - a few questions for you: 
1) For item 1 above, can you clarify this: "It's best if it's created/treated in a special way that allows it to remain easily accessible to the user (during this session or the next) but for the system to know that it has never really been "saved" by the user."  

New File will clear the editor and name the file "untitled#" as you mention above, but then what happens? Does that "untitled#" file show up in the file listing of your current project? If not, how can the file be accessible to the user during that session? And if the file is not saved, how does the user access it during the next session?

2) For item 2 above, it sounds like you want to make Save File perform a Save As. Does this mean we lose normal Save functionality? How does a user save an active file they are currently working on? Would the Save As dialog box pop up each time?

3) On that same note, what would you like the Ctrl-S key do? Save or Save As?

PropGit, 2015-06-10:
@drquinn 
Good questions.

1)  Both the BSE and PT software allows new (and unsaved) source code to exist, be edited, accessed (during that session), compiled, and downloaded.  From the application's and tokenizer/compiler's perspective, this new-and-never-saved source does not exist anywhere in the file system; it only exists in the application itself.  The application delivers the source right to the tokenizer/compiler through memory, rather than via some kind of file system reference.

If I understand correctly, Chrome IDE also delivers the source code right to the tokenizer through memory and doesn't require the user to save any new edits to the file system before tokenizing/downloading.  This means part of the vital work has already been done to support code in a similar fashion as BSE and PT.

The missing piece is that Chrome IDE currently doesn't have a way to navigate between source code except to use the project/file view.  This is something I'd like to change (and enhance) and I've started describing it in Issue #91.  Without this UX enhancement, I can see that this issue puts us in a bind.

> "New File will clear the editor and name the file "untitled#" as you mention above, but then what happens? Does that "untitled#" file show up in the file listing of your current project? If not, how [...]

For now, let's treat it like this:
  * New File clears the editor and names the source "untitled#".
  * No file is actually saved in any visible way (in the project/file view, or in the file system) but the app maintains it via its normal indexdb storage; allowing edits, undos, redos, etc.
  * Since there's currently no way to navigate back to new-and-never-saved source, if the user tries to navigate away from the source, first prompt the user to save the source (ie: "Source code 'untitled#' needs to be saved." with the Save As dialog).
  * If instead, the user simply closes the app, all is well... it simply starts the next session with the new-and-never-saved source open where it left off.

>2) For item 2 above, it sounds like you want to make Save File perform a Save As. Does this mean we lose normal Save functionality? How does a user save an active file they are currently working on? Would the Save As dialog box pop up each time?

Yes and no.  Save File should behave like a normal Save operation except when the source code is new-and-never-saved... in that case, Save File should actually activate the Save As function.  Once the source is saved for the first time, any additional activations of Save File will save over the existing file.

Save As is a separate function, Issue #108, that we haven't assigned a short cut key combination to.

>3) On that same note, what would you like the Ctrl-S key do? Save or Save As?

It should always activate Save... but the Save operation should first check to see if the source has ever been saved, and if not, it activates a Save As in lieu of itself.

PR #156 (2015-06-18): File new behavior
#### What's this PR do?
Changes logic for new files and saving files. New files are now created as blank pages and named with untitled# (with # being the next number after the highest numbered untitled file in your working directory).

Unsaved files will Save As, prompting a Save As overlay with Untitled# showing in the filename field. 
Saved files will simply save to their current name.

Ctrl-S and the save button should work for both of these circumstances.

#### Where should the reviewer start?
npm install (to get Frylord updates)
npm run build

#### How should this be man [... cut]
closed 2015-06-18 by 2fast2fourier