parallaxinc/Parallax-IDE #116: Filter out all 'cu' ports

PropGit, 2015-06-02:
Remove (filter out) all 'cu' type ports from the COM port list any time the list is presented to the user.

phated, 2015-06-02:
Should this also address Bluetooth ports?

PropGit, 2015-06-03:
For now, yes.  We do have some people that use a bluetooth device to program the BASIC Stamp wirelessly, but it's rare.  We'll readdress the filtering scheme again later to allow the user more control over this.  For now, just filter out all 'cu' and bluetooth ports.

2fast2fourier, 2015-06-03:
I think we can safely ignore the `Bluetooth-Incoming-Port` and `Bluetooth-Modem` OSX virtual devices, they shouldn't ever represent a BS2 and can cause a delay when opening the port. Bluetooth serial devices (SPP) should come up under a different name, I'll verify this with a BT serial adapter I use for MCU comms.

PropGit, 2015-06-03:
That sounds good.

phated, 2015-06-03:
Closed by https://github.com/parallaxinc/ChromeIDE/pull/135

closed 2015-06-03 by phated