parallaxinc/Parallax-IDE #117: Add firmware rev column to the device information in port list

PropGit, 2015-06-02:
Add microcontroller's firmware revision information (in a separate column) to the port list when doing identification and downloads.

urbantumbleweed, 2015-06-03:
Development on this issue is pending receipt of documentation.

PropGit, 2015-06-04:
The documentation describing this is page 6 of the [BASIC Stamp Programming Protocol](https://drive.google.com/a/parallax.com/file/d/0B0tev9Y3WaL4WUdaM1pZb29uWW8/view?usp=sharing) in the _Identification Sequence_ section.  You have to read the steps to see where it is; it's not specifically spelled out from the perspective of finding the firmware version number.

__NOTES__
* I wish this were implemented in a cleaner way, but it isn't and we just have to deal with it.
* The Firmware Version is a different part of the sequence for the BS2 module than it is for all the modules that followed it (BS2e, BS2sx, etc.).
* The document is outdated; we've updated firmware on many of the modules since this was written, but the BS2e and later modules follow the same nature where their response is both an identifier and a firmware version indicator.  For example, for a BS2e, it responds with a "e" to indicate "Yes, I'm a BS2e" and _also_ to indicate "My firmware version is 1.0."  But if a BS2e responds with an "f", that means it's a BS2e with firmware version 1.1.  A "g" is firmware 1.2.  The other modules have a different set of responses that follow a similar pattern, but using different letters.
* The set of supported module types and firmware versions that the app must be able to __identify__ is slightly different than the set of modules the app supports __downloading__ to.
  * We always make the software able to identify any previous, current, and the next 1 or 2 possible versions, but it can really only support downloading to the official previous and current versions (and not the next two possible versions) because we can't guarantee that any future module firmware versions will still use the exact same download protocol sequence.  We don't want the software to try to download to a module who's firmware version was invented _after_ the software was released. 
* __[For Identification]__ The Chrome IDE needs to properly identify the modules and recognize the firmware versions listed below:
  * BS2 - firmware versions 1.0
  * BS2e - firmware versions 1.0 through 1.4
  * BS2sx - firmware versions 1.0 through 1.5
  * BS2p (24 & 40) - firmware versions 1.0 through 1.8
  * BS2pe (24 & 40) - firmware versions 1.0 through 1.9
* __[For Downloading]__ The Chrome IDE needs to allow and support the following:
  * BS2 - firmware versions 1.0
  * BS2e - firmware versions 1.0 through 1.2
  * BS2sx - firmware versions 1.0 through 1.3
  * BS2p (24 & 40) - firmware versions 1.2 through 1.7
  * BS2pe (24 & 40) - firmware versions 1.0 through 1.6
* Notice that the software is not allowed to download to versions 1.0 and 1.1 of the BS2p.  That's because the download process was originally different for those versions and they were released to special customers while we were prototyping.

In the BASIC Stamp Editor, I handle "hello codes" and responses for BS2e and above as arrays of single-byte characters, or arrays of sets of characters:
```
  HelloCode : array[BS2E..BS2PE] of char = ('e','X','P','I');
  {Note: the below codes reflect BASIC Stamp response codes that relate to a firmware version number; just because they are listed here does not mean that version is actually supported, but rather, the ModuleVersion array contains supported version numbers.}
  ResponseCode : array[BS2E..BS2PE] of set of char = (['e','f','g','h','i'],['X','Y','Z','[','\',']'],['P','Q','R','S','T','U','V','W','X'],['I','J','K','L','M','N','O','P','Q','R']);
```
Something good to note is the response for later firmware versions of the BS2sx and also the fact that the responses are given as uppercase letters for the BS2p and BS2pe, but the response may actually be upper or lower case which indicates that it's a 40-pin or a 24-pin module, respectively.  The source code that actually checks against the response code array does the footwork to check for upper and lower case responses that match the right letter in the set.

In the BASIC Stamp Editor, I handle checking to see if the firmware version is one it's allowed to download to with a single-dimensional array of sets where the index (TStampMode) is an enumerated value that matches the module type (BS2e, BS2sx, etc).  In the code below (10) means v1.0, and (17) means v1.7, etc.:
```
  {Firmware versions supported.}
  ModuleVersion  : array[low(TStampMode)..high(TStampMode)] of set of char =
       ({None}  [chr(00)],
        {BS2}   [chr(10)],
        {BS2e}  [chr(10), chr(11), chr(12)],
        {BS2sx} [chr(10), chr(11), chr(12), chr(13)],
        {BS2p}  [chr(12), chr(13), chr(14), chr(15), chr(16), chr(17)],
        {BS2pe} [chr(10), chr(11), chr(12), chr(13), chr(14), chr(15), chr(16)]);

```

2fast2fourier, 2015-06-04:
Device scan now exposes `version` field (may be null), we just need to add a column to the device list to show that field. Also version bump `bs2-serial` to `^0.7.3`.

Reassigning to @drquinn for these changes.

PR #140 (2015-06-04): UI Updates for Download Modal and Line Number Margin
#### What's this PR do?
Updates the download modal UI headings and button names. #119 
Pulls in version information on the download modal and displays in the device table. #117 
Fixes the gutter margins for line numbers. #111 
Syntax highlighting changes as requested. #109 

#### Where should the reviewer start?
Load Chrome IDE with changes from this PR.

#### How should this be manually tested?
Open the download modal. Look at the table headings and button names to see that they match the request in #119. Connect a Stamp and check that the  [... cut]
closed 2015-06-04 by phated