parallaxinc/Parallax-IDE #120: Tab key indents selected block

PropGit, 2015-06-03:
Implement multi-line selection indention via tab key.
* Tab - Indent block (Inserts spaces to align with absolute multiples of 2)

phated, 2015-06-18:
@PropGit do you have a list of items you want tackled as part of this? or bugs you have found with indent?

PropGit, 2015-06-19:
@phated Yes.
__BUG__ Selecting a block of text and then pressing Tab replaces all of selection with "tabbed" whitespace.

__Example__

__Selected text:__
[image]

__Tabbed result:__
[image]

__Expected result:__
[image]

closed 2015-06-30 by phated