parallaxinc/Parallax-IDE #121: Shift-Tab outdents selected block

PropGit, 2015-06-03:
Implement multi-line selection outdention via Shift-Tab key combination.
* Shift-Tab - Outdent block (Deletes spaces to align with absolute multiples of 2)

phated, 2015-06-18:
@PropGit do you have a list of items you want tackled as part of this? or bugs you have found with outdent?

PropGit, 2015-06-19:
@phated Yes.
  __BUG__ Selecting a block of text and then pressing Shift-Tab removes _all_ whitespace, slamming the poor text ferociously into the left margin instead of removing just the previous tab position's-worth of whitespace.

  _Example_

  __Selected text:__
[image]

  __Shift-Tabbed result:__
[image]

  __Expected result:__
[image]

closed 2015-06-30 by phated