parallaxinc/Parallax-IDE #128: F6 or Ctrl-I - Identify BASIC Stamp firmware.

PropGit, 2015-06-03:
* F6 or Ctrl-I - Identify BASIC Stamp firmware.

Note: This should display the Identification dialog and activate the scanning process for all ports.

PropGit, 2015-06-17:
Does this need to be broken into multiple issues?  There's also a UX/UI element to activate the Identify function, but it was not specified here.

phated, 2015-06-17:
@PropGit I think that would be good.  Would you like me to do that now?

PropGit, 2015-06-18:
Yes please.  From now on, I'll try to enter issues with a finer grain.  I'm used to thinking of these things as so interrelated that they don't make sense to me as separate items, but I think I understand your perspective now.

phated, 2015-06-18:
@PropGit it seems that there are only 2 issues remaining to this functionality (the UI element and the key combination).  Do you agree?  I've split those 2 issues out.

PropGit, 2015-06-18:
Yes, as long as the Identify function already exists as a separately-callable function, and not as an intrinsically integrated part of the download function.  ie: Download should call Identify as the first part of it's operation, which means that the overlay activated by #157 and #158 can call Identify (also) but then simply update the display and wait for user input (to Identify again or close the overlay).

urbantumbleweed, 2015-06-19:
This issue is being broken up into two smaller, but related issues.  One to create key shortcut and another to create a UI component.  Issues are #157 and #158

closed 2015-06-19 by urbantumbleweed