parallaxinc/Parallax-IDE #129: F7 or Ctrl-T - Perform a syntax check on the code

PropGit, 2015-06-03:
* F7 or Ctrl-T - Perform a syntax check on the code.

This should tokenize the code and show the results, just like what happens at the beginning of a download sequence.

urbantumbleweed, 2015-06-18:
I was hoping to get a little more clarity by what is meant by 'show the results'.  When syntax check is run independently, what exactly should the user be seeing when:
* all the syntax is correct?
* when there is a syntax error?

PropGit, 2015-06-18:
* When all syntax is correct, a temporary, green toast should appear saying "Tokenization successful!"
  * When there is a syntax error, a red toast should appear showing the tokenizer's error message (stripped of the error number and the "-" following the number) and the editor should ensure the offending line is visible and the selected offending text characters are selected (highlighted).

closed 2015-07-02 by phated