parallaxinc/Parallax-IDE #133: Bundle example files with ChromeIDE

PropGit, 2015-06-03:
[This is a placeholder for bundling example source code files with the package]

urbantumbleweed, 2015-06-18:
I am wondering if there is a design element to this?  For example, are the files going to structured as a type of tutorial?  If so, are we going to introduce application functionality or will the emphasis be on PBASIC and programming.  It seems like final collection of files will be the result of some user testing.

PropGit, 2015-06-18:
These are BASIC Stamp source code files with various file extensions (.BS2, .BSE, .BSX, .BSP, and .BPE) that need to be stored in a group which is accessible via the Project/File view.  At the most, we may ask for an "Examples" menu item to be added for quick access, but I'm _not_ doing so right now.

We do have tutorials in the form of .PDF and possibly .HTM files that we'd like to include in the future, but this will be handled as a separate issue that will likely have some kind of a UX/UI element to it.

urbantumbleweed, 2015-10-08:
@propgit: There are a number of decisions that must be made before development can begin. You can use the checkboxs to indicate what tasks/questions are completed/resolved.

- [ ] Step 1: Answer discovery questions and define specifications.
  - [ ] The actual content to be contained in the examples folder has been created.
  - [ ] What interaction (in regards to the example folder) is expected the first time the application is run?
  - [ ] What information (if any) about the examples will be available in the help view?
  - [ ] What should happen when a user alters an example file and attempts to save?
  - [ ] How should a user revert to the original state of the entire example folder?
  - [ ] How should a user revert to the original state of an individual example file?
  - [ ] Will is be possible for user to delete or override the example folder?
  - [ ] If applicable, how should a deleted examples folder be restored?
  - [ ] Will it be possible for the contents change overtime, possibly by an educational administrator or course instructor?
- [ ] Step 2: Audit specifications with UXUI designer.
- [ ] Step 3: Create Github issues that are related to this issue.
- [ ] Step 4: Incorporate into development sprints.
- [ ] Step 5: Example folder and content implemented.

PropGit, 2015-10-08:
@urbantumbleweed - great idea with the checkboxes!
I can answer some of these now; others need more work on my part.

>What information (if any) about the examples will be available in the help view?

You mean in the on-line help link?  Likely none (or very little) initially.  We will probably just mention the examples and how to get to them.  In later revisions of the on-line help, we'll likely more directly refer to certain examples (which we might expand on later) that naturally go with certain tutorial information contained in the help system.

>What should happen when a user alters an example file and attempts to save?

Ideally, we'd like it to be read-only and instead the app could automatically recognize this is the examples project and can suggest to the user that they save their modification to another place.  However, in previous discussion you said that "read-only" was very hard to do.

So, it should just let them save over the example file as normal (no code changes needed in Parallax IDE).

>How should a user revert to the original state of the entire example folder?

If the app can recognize and treat the example folder as something special; we could provide a UI/UX feature to restore the examples.

Regardless, if the user deletes the example folder, or otherwise renames it, upon next session, the app should see that it is missing and recreate the entire thing as it did on the very first session after install.

PropGit, 2015-10-11:
>What interaction (in regards to the example folder) is expected the first time the application is run?

The first time the application is run, after the "updated" prompt that already appears, the app will notify the user via overlay that an examples project has been provided for them.

PropGit, 2015-10-13:
> The actual content to be contained in the examples folder has been created.

The _first_ version of examples folder content is here for this release: [Examples Project Files](https://drive.google.com/file/d/0B0tev9Y3WaL4Y0JkSVZZRDhVUFE/view?usp=sharing)

PropGit, 2015-10-13:
> How should a user revert to the original state of an individual example file?

Ideally, the app could recognize that this is a special "examples" folder and provide a UI/UX feature (ex: right-click > restore example) to restore it.  

For now, this is not a feature we will implement.  Instead, the only way to restore will be by deleting and restoring the entire examples folder.

> Will is be possible for user to delete or override the example folder?

Yes, they can delete the folder (which is one way to lead to restoring it) and then can recreate it and fill it with files; but the future app may still see and treat that folder as a special _examples_ folder.

> If applicable, how should a deleted examples folder be restored?

Answered at bottom of [this post](https://github.com/parallaxinc/Parallax-IDE/issues/133#issuecomment-146680618) - the app recognizes it's missing upon next session and restores it.

> Will it be possible for the contents change overtime, possibly by an educational administrator or course instructor?

Yes, by Parallax staff.  We'd need to update the repository's contents so next build includes new examples.  At this time, we'd like it built into the app, rather than provided via some Internet resource.

PropGit, 2015-10-13:
The implementation of this feature will be done in phases, across at least two sprints.  Other issues describe the separate related features.

PropGit, 2015-10-13:
Since other issues were created to resolve this, this issue can be closed in this sprint after it is read and understood.

urbantumbleweed, 2015-10-14:
@PropGit @phated Here is a listing of issues that I think can be created from this thread.  I have included the two issues that were created already (#295 and #296)

 1. Ensure that 'Examples' project exists on session start (#295)
 1. Create UX/UI to restore examples (#296)
 1. Online help  refers to certain examples (TBD)
 1. App automatically recognizes the current folder is the is the examples project and can behave differently.
 1. Saving an example file copies the content state, creates a new file, reverts the example file, and moves to the new file.
 1. Provide the option to hide the examples folder.
 1. Create codemirror documents and example directory from zip file [here](https://drive.google.com/file/d/0B0tev9Y3WaL4Y0JkSVZZRDhVUFE/view?usp=sharing)
 1. Pull the content from an internet resource.

Please confirm that these are the issues that we should create.  If so, let's add them so that in subsequent sprints we can add them.

phated, 2015-10-14:
@urbantumbleweed it looks like a bunch of these would be enhancements that wouldn't get implemented until a ways into the future.  Can you mark those in your list in some way?

urbantumbleweed, 2015-10-15:
Updated, let me know what you think.

PropGit, 2015-10-15:
@urbantumbleweed @phated 
Number 3 - seems to provide a pseudo-read-only feature.  Is that right?  The intent of this item is to preserve the original example?  This begs the question, what happens with unsaved changes?

Number 4 - I don't understand this one.  Please explain.

urbantumbleweed, 2015-10-15:
Shoot, I moved the issues around and don't know which is Number 4.  Please let me know which one it is.

For Number 3, it is the pseudo-read-only feature.  If there are unsaved changes then the original example file would remain intact.

PropGit, 2015-10-15:
The one I don't understand is "Create CodeMirror documents and example directory from zip file here"

I don't know what CodeMirror documents are or why they apply here, and the example directory is what is being created in Issue #295.

Other than that, I like the others and agree that we should make issues out of them.  Would you like me to create them?

urbantumbleweed, 2015-10-16:
All that was meant was that we would use the content from the supplied zip archive to create the example files in the application.  I have rephrased the issue above because we won't be creating CodeMirror documents but will be creating files on the application file system.

Yes, please create the issues at your convenience.

closed 2015-10-16 by urbantumbleweed

PropGit, 2015-10-16:
I created a number of related issues as requested and also updated [the list above](https://github.com/parallaxinc/Parallax-IDE/issues/133#issuecomment-148231456) to reference them.