parallaxinc/Parallax-IDE #136: Unplugging + plugging in device causes error upon download without refreshing device list

phated, 2015-06-03:
@PropGit will this still be a problem when we refresh the device list every time the download dialog is opened?

PropGit, 2015-06-04:
For downloads, no, it won't be a problem because the next download will refresh the list and then will automatically select the port that it found the target BASIC Stamp on.

The point is to make the app assess the situation and if it's not ambiguous, it should just automatically do the things the user would have done anyway... ie: source code is tokenized for a BS2 and only one BS2 plugged in?... select that module and download to it without any further input from the user.

phated, 2015-06-18:
I believe this has been discussed and answered sufficiently.  Closing as it shouldn't be an issue.

closed 2015-06-18 by phated