parallaxinc/Parallax-IDE #138: Activate port enumeration and scanning automatically upon Download and Identification functions

PropGit, 2015-06-04:
When the user selects the Download or Identify functions, once the progress dialog appears, make the app automatically begin the port enumeration (determining which ports exist) and port scanning (performing the Identification part of the BASIC Stamp Programming Protocol) and update the dialog's port listing with the results.

phated, 2015-06-17:
@PropGit was this completed on the last sprint?  If not, how does it differ from what we currently have?

PropGit, 2015-06-17:
@phated Wow, I don't know how I did that, but you're right this is a duplicate of Issue #74 and Sprint 4 implemented it.

This issue can be closed and what really needs to happen next is Issue #139 which is vital to creating a single-step action for the user to perform a download (issue #131).

closed 2015-06-17 by PropGit