parallaxinc/Parallax-IDE #139: Automatically determine target stamp after port scanning and download

PropGit, 2015-06-04:
If the user selects the Download operation, after the progress display updates with the results of the port enumeration and port scanning processes (described in issue #138), the app should now determine the target Stamp to download to.

Decision Flow:
  1. No modules of any kind are found to be connected?
    * Leave progress display on-screen (for user review)
    * Display error: "No BASIC Stamps found."
    * NOTE: the intent is for the user to be able to read the error and view the list of ports scanned to help them determined what was done and what the results were
  2. Only one instance of the source code's specified module type is found to be connected to the system?
    * Automatically select that module type's port and perform the entire programming protocol on it.  This means reset, identify, program, verify.  Yes, the module will actually have been identified twice and programmed once through the processes of issues #138 and #139.
  3. More than one instance of the source code's specified module type is found to be connected to the system?
    * Prompt the user to select the module they wish to download to; "Please select which module to download to."
    * If the user selects the Cancel button; close the display and cancel operation
    * If the user selects a module, immediately perform the entire programming protocol on it.  This means reset, identify, program, verify.  Yes, the module will actually have been identified twice and programmed once through the processes of issues #138 and #139.
      * NOTE: It's not desired, but if there _must_ be a Download button that they have to click after selecting the desired module, it's acceptable.  We'd like to avoid forcing this extra click, however.
  4. No modules of the type specified in the source code are found to be connected, however, at least one module of another type were found to be connected?
    * Prompt user with "BASIC Stamp X not found.  Select alternate module instead?" ;where 'X' is the type of BASIC Stamp that the code was written for.
    * If the user selects the Cancel button; close the display and cancel operation
    * If the user selects a module,
      * automatically modify the source code to change the $STAMP directive to reflect the new desired module type (you can use the tokenizer's returned information to find the specific source and location to change)
      * immediately re-tokenize the source code (for that new module)
        * If tokenization error, display error and cancel the rest of this download process
      * if no tokenization error, perform the entire programming protocol on the module that the user selected.  This means reset, identify, program, verify.  Yes, the module will actually have been identified three times and programmed once through the processes of issues #138 and #139.

urbantumbleweed, 2015-06-18:
I am labeling this as a UX issue because of the prompting in point 3.

phated, 2015-06-18:
@urbantumbleweed I think you are incorrect in labeling this as such

urbantumbleweed, 2015-06-18:
In response to @phated regarding this as a development issue, the UX/UI label has been removed.

phated, 2015-06-18:
@PropGit I think we should split this into multiple, related issues.  Do you agree?

PropGit, 2015-06-19:
You mean one issue for each of the four possibilities in the decision flow?

If it helps get the the work done, yes.  These are all part of one function really; they shouldn't be developed separately without thought of the other.  1 & 2 really need to be done at once.  3 & 4 can be developed later, but when 1 & 2 are made they need to set aside space for 3 & 4.

phated, 2015-06-19:
@PropGit I think 1, 2 and 3 are all able to be done at once, but 4 will be done at another time.

PropGit, 2015-06-19:
@phated Okay, yes, 4 is the most involved of any of them.

urbantumbleweed, 2015-06-19:
The issue has been split out into multiple issues.  For more information take a look at the individual issues #163, #164, #165, #166

closed 2015-06-19 by urbantumbleweed