parallaxinc/Parallax-IDE #151: Changing Name from "Chrome IDE" to "Parallax IDE"

PropGit, 2015-06-17:
We are officially changing the name of this project as the subject says.

This means
  * We need to change occurrences of "Chrome IDE" in the application and the repository sources (like readme.md and the releases' QuickStart Installation notes) to "Parallax IDE"
  * I will be changing the repository soon.

NOTE: historical items (like previous release archives) do not need to change; they can remain as is, but future such items should carry the new name.

PropGit, 2015-06-17:
I can change the repository name right now if you're ready.  I'll wait until I see a response.

phated, 2015-06-18:
@PropGit I think we are ready for you to change the name

PropGit, 2015-06-18:
Okay, done!  The official name is "Parallax IDE" but the repo is "Parallax-IDE"

phated, 2015-06-19:
@PropGit it seems you need to change the Travis-CI hook now, as it is trying to get from ChromeIDE still.

PropGit, 2015-06-19:
@phated Sorry, need help on this one.  Can't find how I did this the first time and if I log into Travis, it thinks its the first time I've been there.  Email me if need be.

phated, 2015-06-19:
@PropGit I started making a series of images to walk you through this, but it seems that Travis-CI has already done the updating behind the scenes.  I didn't know they started doing that.

PropGit, 2015-06-19:
Is Travis CI working now?

phated, 2015-06-19:
@PropGit yep, it sorted itself out! :astonished:

commit 84c3f850 (2015-06-19): created new images with consistent references and better highlighting. Closes #151
closed 2015-06-19 by urbantumbleweed