parallaxinc/Parallax-IDE #152: Save As - Save a file as a different name

phated, 2015-06-17:
Split out from #108

drquinn, 2015-06-24:
@PropGit when Save As is triggered, what would you like to show in the filename box?  Currently it will show the active file name (see image below for file "a test"). Does that work?  Or would you like it to show "activeFilename copy"? Or "activeFilename1"? Or continue on with the "untitled#" scheme? Something else?

[image]

jeffam70, 2015-06-24:
@drquinn Thanks for asking.  Since the common workflow is for someone to want to _save as_ to a similar (but different) name, I'd like to to show the active name (as you now have it) with the field focused and with the cursor to the right of the name.

Also, this reminds me of a needed feature... When the user clicks the _Save As_ button from this overlay, the Save As feature should always prompt with a confirmation if the user uses a filename that already exists (ie: "File '<filename>' already exists.  Overwrite anyway?" with _Overwrite_ and a _Cancel_ button options).

I can make that a separate linked issue if you want.

drquinn, 2015-06-24:
@jeffam70 ok great, I'll set that up for filename.

I think another issue would be good for the overwrite prompt and dialog that would go along with it.

closed 2015-06-30 by phated

PropGit, 2015-07-27:
Tested, but the cursor doesn't end up to the right of the filename.  Will make another issue for this.

PR #169 (2015-06-30): add saveAs action with keypress
#### What's this PR do?
Adds the ability to Save As from any file using the keypress CTRL-SHIFT-S.

#### Where should the reviewer start?
Load the branch.
npm run build

#### How should this be manually tested?
Open a file with some text inside. Ensure you can save with CTRL-S or by clicking the Save button.

Next try to Save As with CTRL-SHIFT-S. You should get a dialog where you can enter a new filename, then press Save As in the dialog. The file should save and navigate you to the new file. 

#### What are the relevant tickets?
Closes #152 
Closes #155