parallaxinc/Parallax-IDE #166: Automatic Download - No module types that match but alternative alternative module available

phated, 2015-06-19:
__Scenario__
No modules of the type specified in the source code are found to be connected, however, at least one module of another type were found to be connected

__Behavior__
* Prompt user with "BASIC Stamp X not found. Select alternate module instead?" ;where 'X' is the type of BASIC Stamp that the code was written for.
* If the user selects the Cancel button; close the display and cancel operation
* If the user selects a module,
    * automatically modify the source code to change the $STAMP directive to reflect the new desired module type (you can use the tokenizer's returned information to find the specific source and location to change)
    * immediately re-tokenize the source code (for that new module)
        * If tokenization error, display error and cancel the rest of this download process
    *if no tokenization error, perform the entire programming protocol on the module that the user selected. This means reset, identify, program, verify. Yes, the module will actually have been identified three times and programmed once through the processes of issues #138 and #139.

drquinn, 2015-07-14:
@PropGit quick question about finding the $STAMP directive in the user's source code. What are the fields that matter in the PBASIC tokenizer response that I need to look at to figure out where that directive is?

PropGit, 2015-07-15:
Sorry, I'm away until late tonight.

Jeff Martin
Sr. Software Engineer, Parallax Incorporated
(916) 625-3002
jmartin@parallax.com
http://www.parallax.com

phated, 2015-07-15:
@PropGit that is fine.  We are working on the Transmit Pane right now and just wanted to get this question posted to you as early as possible.  Thanks.

PropGit, 2015-07-15:
@drquinn the tokenizer fills in the TargetModule and TargetStart fields when compiling the code.  The TargetModule is a value 0 through 6 indicating the type of Stamp the code is targeting and the TargetStart indicates the character position within the source that the target module string starts.  The TargetStart treats the source like one long continuous string, where LF or CR is just another character rather than the end of a line.

The notes on this can be found in the [Using the PBASIC Tokenizer Library v1.2.pdf](https://drive.google.com/file/d/0B0tev9Y3WaL4SFF2c0NHdDZKd00/view?usp=sharing) document, page 5.  What I don't know is whether or not the wrapper API that Iced Dev made actually exposes both these fields or not.

So
```
'{$STAMP BS2}
'{$PBASIC 2.5}
```
would report a TargetModule of 2 and a TargetStart of 9.  For purposes of resolving this issue's feature, you have to know that 2 equates to the 3-character name "BS2" to remove those characters from the source (starting at character position 9) and replace them with the appropriate module identifier (which could be up to 5 characters in length; BS2, BS2e, BS2sx, BS2p, BS2pe).

If on the other hand the source was this
```
'{$PBASIC 2.5}
'{$STAMP BS2}
```
the tokenizer would report a TargetModule of 2 and a TargetStart of 24 (assuming the source uses single-character line endings).

closed 2015-07-17 by 2fast2fourier

PropGit, 2015-07-24:
Verified Added!

PR #193 (2015-07-17): Automatic download no match
#### What's this PR do?
Rewrites the source code if no matching BS2s are available but a different BS2 is and is selected.  Then recompiles the source and programs the device.

#### What are the important parts of the code?
`src/stores/documents` needed an `update` method and to track the current filename so we could alter the source code without causing a change event on CodeMirror.  `src/stores/device` is where the source code replacement is happening.  Not sure if that is the best place for it but it works for the current implementation.

#### How should this be  [... cut]