parallaxinc/Parallax-IDE #168: Save As - Overwrite Protection (Prompt)

jeffam70, 2015-06-24:
As it relates to Issues #152 and #153 (Save As overlay), add a feature that protects against accidental overwriting.

On the Save As overlay, when the user clicks the Save As button, the system should always prompt with a confirmation overlay if the filename matches an already existing file in the target folder.  The prompt should say, "File 'filename' already exists. Overwrite anyway?" and should have an Overwrite and a Cancel button.

phated, 2015-06-30:
I think this is a bit too much to add to this sprint.  Let's slate it for the next sprint

PropGit, 2015-07-01:
Okay.

urbantumbleweed, 2015-09-10:
@monteslu to address this issue you are going to want to look at the following files:

1. [src/views/save-overlay.js](https://github.com/parallaxinc/Parallax-IDE/blob/master/src/views/save-overlay.js)
1. [src/plugins/overlays.js](https://github.com/parallaxinc/Parallax-IDE/blob/master/src/plugins/overlays.js)
1. [client.js](https://github.com/parallaxinc/Parallax-IDE/blob/master/client.js)

I suspect that the actual logic will go in src/views/save-overlay.js

closed 2015-09-14 by phated

PropGit, 2015-09-16:
Verified done.

PR #264 (2015-09-14): adds confirm dialog on overwrite
#### What's this PR do?

adds a confirmation overlay when attempting to overwrite an existing file

#### Where should the reviewer start?

the logic change starts in the handlers.js

#### How should this be manually tested?

pull down, build, load, then create a new file.  Attempt to save it with the name of an existing file.
Also, save a new file with a name that doesn't exist yet and dialog should not appear.

#### What are the relevant tickets?

fixes #168