parallaxinc/Parallax-IDE #176: Need release notes/changelog for 0.4.0 and 0.5.0

phated, 2015-07-03:
We had great release notes in all releases until 0.4.0.  Please update the release notes for 0.4.0 and 0.5.0 releases.

urbantumbleweed, 2015-07-15:
@PropGit It looks like the documentation for 0.4.0 and 0.5.0 have been updated and the issue can be closed.  Will you confirm that it is ok to close this issue?

PropGit, 2015-07-16:
Hi @urbantumbleweed.  Yes, I updated those myself and forgot to close this issue.  I've decided that I'd like to write those release notes myself from now on because we're using that release page as a source for customers to visit when they want to know what has changed in the latest version.  So, it's okay with me if you leave it blank for future releases, or if you add a few notes that you don't mind me touching up a bit to help users understand.

closed 2015-07-16 by PropGit

phated, 2015-07-16:
@PropGit sounds good!  Great work on the latest ones, I really like the linked issues.

urbantumbleweed, 2015-07-16:
Not a problem at all. I too really liked the linked issues and will do that
going forward when I can.

PropGit, 2015-07-16:
Perfect.  Thanks guys!