parallaxinc/Parallax-IDE #187: Can Chrome Apps do X?

PropGit, 2015-07-10:
Can a Chrome Application...
  * Determine the OS it is running on?
  * Fetch a downloadable resource from a URL?
  * Automatically execute that resource?
  * Provide user prompt for action that then executes that resource?

I'm asking because we could greatly improve user experience if the app itself could determine if the user's system needs a manually-installed USB driver and provides and installs that driver automatically, or semi-automatically.

urbantumbleweed, 2015-07-16:
@propgit: I have researched this issue over the past couple days and thought I would bring a summary of what I found to the thread.

In regards to whether the Chrome App can determine the OS it is running on:
The Chrome App can determine the OS it is running on.  Chrome has a `runtime` API that provides us this information.

In regards to to downloading, automatically executing, and/or prompting the user for install:
These additional questions require further explanation and present a couple options.  Chrome does provide a `downloads` API which does everything we need, except that it is only permitted for use in extensions, not packaged apps (chrome apps).  There is a related [chromium issue](https://code.google.com/p/chromium/issues/detail?id=255554) and [google group discussion](https://groups.google.com/a/chromium.org/forum/#!topic/chromium-apps/O-qk197U36Y) that has was active a month ago which presents an alternative to the preferred downloads api using something called a ‘webview’.  This would allow us to download external content from within Parallax IDE.  For more information the webview you can go to the [google developer docs](https://developer.chrome.com/apps/tags/webview).

We are presented with three options:
1. Build a chrome extension that utilizes the chrome.downloads
2. Utilize a <webview> tag in Parallax IDE
3. Notify the user with a link to the drivers

**Let’s discuss each one:**

**Build an extension/app combo:**
Because Chrome Apps can easily communicate with a Chrome Extension it is our recommendation that we build a Chrome Extension that has the exclusive purpose of downloading and installing drivers.  This is option is the most robust and gives us the greatest flexibility. This makes sense for us because it separate concerns in a clear way. The communication API will allow the Chrome App to use the extension to download, open and install the drivers as needed.  Or, if not automatically install, then it could prompt the user to install.  If we decide to take this approach, then we need to discuss how we facilitate installation and use for the user.  One possible drawback is that it will require two items to be installed.  On a positive note, there are applications outside of Parallax IDE where the extension would be a great utility.

**Use \<webview\> element:**
The second option is to use a \<webview\> element in Parallax IDE.  The <webview> element acts somewhat like an \<iframe\> element with the difference being that it uses a separate thread from the main process and operates asynchronously for security reasons.  This means that we can access external content and style that element according to the theme of the editor.  The limitation here is that it does not allow for inline execution.  The user will still need to go to downloads and open the executable manually.  This is perhaps simpler to implement than the extension/app combination, but presents other issues.  For example, the user may need permission to save or open the downloaded file because it is not being performed programmatically.  Another drawback is the extent to which the user needs to be involved in the process.

**Simple Notification:**
The third option is to simply perform a check and then present the user with a link to the appropriate resource, either as a banner, overlay, or toast.  This would be the simplest to implement but will be the most manual process for the user.  For this approach to be successful, we must also ensure that a user will have the proper permissions at each step to complete the install.

In conclusion, it is possible for us to identify the host OS, initiate a driver download if needed, and programmatically install the drivers.  The APIs that make this possible are not directly available to Chrome Apps but can be used directly by Chrome Extensions which can communicate with Chrome Apps. This circumstance presents us with three options; each with advantages and drawbacks.

A good next step for us is to discuss user expectations and which option would best fulfill the needs.

For your reference, here are some of the resources I looked at during my research.
[chrome.downloads](https://developer.chrome.com/extensions/downloads)
[google groups discussion](https://groups.google.com/a/chromium.org/forum/#!topic/chromium-apps/O-qk197U36Y)
[webview element](https://developer.chrome.com/apps/tags/webview)
[chromium issue requesting chrome.downloads to work in chrome apps](https://code.google.com/p/chromium/issues/detail?id=255554)

closed 2015-07-16 by urbantumbleweed

PropGit, 2015-07-24:
Is it possible for a Chrome Extension to automatically install a Chrome App?