parallaxinc/Parallax-IDE #195: Holds port open after failure to identify

PropGit, 2015-07-17:
Parallax IDE needs to close the port after a failure to identify or download.  See [video for example](https://drive.google.com/file/d/0B0tev9Y3WaL4UzlGYlB5TnY3UWs/view?usp=sharing) that uses the BSE's Identify feature and Parallax IDE's Identify feature.

This problem may also be preventing Parallax IDE itself from reopening the port upon the next Identify/Download operation.

NOTE: Parallax IDE should have been able to identify during this case, but it also has a problem handling serial data entering the receive buffer before (or right after) the reset condition is issued, as noted by Issue #194.

phated, 2015-08-07:
This should be fixed with dependency updates.

closed 2015-08-07 by phated

PropGit, 2015-08-17:
Verified correct.  Yay!