parallaxinc/Parallax-IDE #204: Save As Overlay: Put cursor to right of current file name

PropGit, 2015-07-27:
Issue #152, in Sprint 6 indicated that [the field should be focused and the cursor should be to the right of the file name](https://github.com/parallaxinc/Parallax-IDE/issues/152#issuecomment-115035841).  

The cursor still appears to the left of the file name upon activation of the Save As overlay.  Please fix it so that the cursor appears to the right of the current file name every time the overlay is activated.

PropGit, 2015-10-01:
__[Bug Introduced]__
The solution employed to solve this issue causes cursor navigation to fail while editing the filename.

Clicking in the field (to place the cursor anywhere but at the end), cursor keys, HOME, Ctrl and/or Shift + cursor key combinations (and more) are thwarted, forcing the user to edit the whole filename only from the right side.  Any cursor movement is undone and the cursor is once again returned to the right side of the filename.

phated, 2015-10-01:
@PropGit I am unable to reproduce what you are explaining.  None of the code I modified would cause it either

PropGit, 2015-10-01:
@phated - Whoa... is this perhaps a Windows thing or a Chrome 45 thing?

Wait.  I built off of the break-compat branch; perhaps that's the source of the problem.  Maybe it's a non-issue for the pre-release you're building today?

phated, 2015-10-01:
@PropGit I am using Chrome 45 on my mac.  I will try to get a windows VM running but the cursor stuff shouldn't be platform dependent.  The break-compat branch had the same changes in it.

PropGit, 2015-10-01:
@phated - I'll make a video then.

PropGit, 2015-10-01:
@phated - Here's [a demo video](https://drive.google.com/file/d/0B0tev9Y3WaL4UW84bXZCLWJJSTQ/view?usp=sharing) of the symptoms I'm seeing.

phated, 2015-10-01:
@PropGit thanks for the video.  I did a fresh install and build and am able to see this behavior now.  Working on a fix before the release.  Thanks for catching this and showing me.

phated, 2015-10-01:
@PropGit I've pushed a fix to the master branch.  Can you please build from master, test and let me know if it is working?  If so, I will cut the release

PropGit, 2015-10-01:
@phated - Yes.  I will do that now...

PropGit, 2015-10-01:
@phated - Success!  It works perfectly now, thank you!

closed 2015-10-01 by PropGit

phated, 2015-10-01:
@PropGit Hooray!  Cutting the release shortly

PropGit, 2015-10-03:
Verified fixed.

PR #285 (2015-09-28): get cursor to be to the right of the text in the save overlay input
#### What's this PR do?
Gets the cursor to the right of the text in the save overlay.  This is achieved by removing the text before focus and then putting it back after (yay web :unamused:)

#### What are the important parts of the code?
The `focusInput` method of `src/views/save-overlay.js` has been modified to do what was described above.

#### How should this be tested by the reviewer?
Try the different methods of opening the save dialog (change file, cmd+shift+s, etc) and verify the cursor is always to the right.

#### Is  [... cut]