parallaxinc/Parallax-IDE #207: Automatic Save Prompt fails for new files after session

PropGit, 2015-07-27:
[This is likely related to Issue #206] 
New file's status is forgotten between Parallax IDE sessions - leading to loss.

  * Create a new file
  * Don't save it
  * Close Parallax IDE
  * Start Parallax IDE
  * [Source is displayed, as expected]
  * Select another file from the Project/File view
  * Parallax IDE immediately displays new file without prompting to save the new and unsaved file first
    * The new and unsaved file is lost forever

NOTE: When this is fixed, make sure the solution doesn't negatively affect the case where the user has modified an existing file, closed Parallax IDE, started Parallax IDE and navigated away to another file- this works perfectly as expected.

phated, 2015-08-07:
Due to the filesystem refactor, this should be fixed in sprint 8.

PropGit, 2015-08-07:
Looking forward to it.

PropGit, 2015-08-17:
Unable to fully test this because of issues #231, #232, #234, #235, and #236.

closed 2015-08-25 by 2fast2fourier

PropGit, 2015-08-30:
Verified fixed.

PR #243 (2015-08-25): update user config module and make sure saving is waited on
#### What's this PR do?
Updates the user config dependency and usage of it.  The new API returns promises for when setting/unsetting the values complete, which should keep us from losing changes to config.

#### What are the important parts of the code?
`src/plugins/handlers.js` has most of the updated API logic.  Notice that each method call is inside an arrow function to handle returning of promises.

#### How should this be tested by the reviewer?
Use the editor, moving between files, creating new files, editing, saving, etc.  Make [... cut]