parallaxinc/Parallax-IDE #208: File System Error - new unsaved file lost between sessions

PropGit, 2015-07-28:
[This happened twice.  May have to do with speed of operations.]

  * Created a new file
  * Typed some text
  * Closed Parallax IDE
  * Started Parallax IDE
    * Error... file lost (see image)
[image]

PropGit, 2015-07-28:
Just retried.  It's possible to make changes in a new file, close Parallax IDE, start Parallax IDE and that file is lost (sometimes without visible error; just a different file in the view instead of the last file from last session).

Can you put a trap upon Parallax IDE close event that causes it to finish storing any unstored changes before fully closing down?

phated, 2015-08-07:
Much of this was probably a side effect of the bad filesystem implementation.  I believe it should get better after this sprint, but might not be completely resolved.

Unfortunately, there is no way to implement a trap in the editor unless **ALL** filesystem operations were deferred to the background page (and I'm not sure if the background page has that access) which would be incredibly inefficient for the editor due to message passing serialization.  Background pages are the only thing guaranteed to be running and able to catch a close event initiated from exiting the application.

closed 2015-08-25 by 2fast2fourier

PropGit, 2015-08-30:
Verified that this seems to be fixed.  It was not a definitive problem (occurred at random) but I haven't seen it happen through extensive tests in v0.7.1 now.

PR #243 (2015-08-25): update user config module and make sure saving is waited on
#### What's this PR do?
Updates the user config dependency and usage of it.  The new API returns promises for when setting/unsetting the values complete, which should keep us from losing changes to config.

#### What are the important parts of the code?
`src/plugins/handlers.js` has most of the updated API logic.  Notice that each method call is inside an arrow function to handle returning of promises.

#### How should this be tested by the reviewer?
Use the editor, moving between files, creating new files, editing, saving, etc.  Make [... cut]